EnviroDIY/ModularSensors #231: Support all sensors used in workshops

SRGDamia1, 2019-02-20:
[] TMP36
[] TLS2561

aufdenkampe, 2019-06-03:
@fisherba and I feel strongly that sensors used in workshops should be ones that participants might actually deploy in a real-world monitoring situation. 

For that reason, we developed the [Learn EnviroDIY Programming](https://envirodiy.github.io/LearnEnviroDIY/index.html) tutorial and related workshops to use:
- a [DS18B20 Waterproof Digital Temperature sensor](https://www.adafruit.com/product/381)
- a [Grove BME280 Temperature, Pressure, and Humidity sensor](https://www.seeedstudio.com/Grove-Temp-Humi-Barometer-Sensor-BME28-p-2653.html)

These are relatively inexpensive and already have wrappers in this ModularSensors library.
- https://github.com/EnviroDIY/ModularSensors/wiki/Maxim-DS18
- https://github.com/EnviroDIY/ModularSensors/wiki/Bosch-BME280

SRGDamia1, 2019-06-03:
I'm almost completely removed from all workshop content.

@darscott @s-hicks2 @ensign@stroudcenter.org