EnviroDIY/ModularSensors #235: missing dependency ADS1X15

neilh10, 2019-03-05:
I'm running into some issues with dependencies and so decided to rebuild 
ModularSensors\examples\menu_a_la_carte\menu_a_la_carte.ino
from scratch. That is .pioenvs and.piolibdeps were deleted.
~/Documents/Arduino/enviroDIY/ModularSensors 
$ git remote -v
origin  git@github.com:EnviroDIY/ModularSensors.git (fetch)
origin  git@github.com:EnviroDIY/ModularSensors.git (push)

$git pull
neilh77@nhpinsp MINGW64 ~/Documents/Arduino/enviroDIY/ModularSensors (master)
$ git log -1
commit cbf64a609d81253abfde72ac7aac07653358a3ec (HEAD -> master, origin/master, origin/HEAD)
-----  (ie latest off)
In PlatformIO/VS Code - 
I press button "platformio build"

which gives
.......
LDF MODES: FINDER(deep+) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Error: Could not find `ADS1X15` dependency for `EnviroDIY_ModularSensors` library
============================================== [ERROR] Took 10.42 seconds ==============================================

neilh10, 2019-03-05:
It looks like there is a name mismatch between library.json and the actual name of the lib when it is pulled in .... something to do with the auto libfinding magic. 
I manually managed to get a copy in.

neilh10, 2019-03-06:
https://community.platformio.org/t/soligen2010-ads1x15-library-disappeared-from-registry/6892

SRGDamia1, 2019-03-06:
The owner of that library has been trying to get Adafruit to accept a pull request from him to update their own buggy library.  To that end, the library.json file was deleted meaning the library disappeared from the PIO registry.  I put it in the dependencies by a direct link to the repo.

neilh10, 2019-03-06:
Just checking - I would expect a fix to the master branch to have been submitted (and referenced) to fix this. I can't find anything to pull. :)

SRGDamia1, 2019-03-06:
Sorry, I just pulled it into develop.  I'll get it up to master soon!

fisherba, 2019-03-21:
This dependency seems to be still missing. 

Error: Could not find ADS1X15 dependency for EnviroDIY_ModularSensors library

I had this error multiple times when compiling simple_logging to a completely fresh directory to test for tutorials.

SRGDamia1, 2019-03-21:
I can't reproduce this.  It's downloading fine for me.

SRGDamia1, 2019-03-21:
What does your platformio.ini look like?

fisherba, 2019-03-21:
I am using it based off of the example for simple_logging: 
```
[platformio]
src_dir = deployments/simple_logging

[env:mayfly]
monitor_speed = 115200
board = mayfly
platform = atmelavr
framework = arduino
lib_ldf_mode = deep+
lib_ignore = RTCZero
build_flags =
    -DSDI12_EXTERNAL_PCINT
    -DNEOSWSERIAL_EXTERNAL_PCINT
lib_deps =
    EnviroDIY_ModularSensors
;  ^^ Use this when working from an official release of the library
;    https://github.com/EnviroDIY/ModularSensors.git#develop
;  ^^ Use this when if you want to pull from the develop branch
    https://github.com/PaulStoffregen/AltSoftSerial.git
    https://github.com/SRGDamia1/NeoSWSerial.git
    https://github.com/EnviroDIY/SoftwareSerial_ExternalInts.git
;  ^^ These are software serial port emulator libraries, you may not need them
```

fisherba, 2019-03-21:
My intent is to instruct people to use the ini that is associated with the example file. I really like that you include an ini example.

SRGDamia1, 2019-05-31:
I think has been solved

closed 2019-05-31 by SRGDamia1