EnviroDIY/ModularSensors #236: Break clock support into separate module

SRGDamia1, 2019-03-06:
Current supporting DS3231 via EnviroDIY DS3231 library and SAMD internal clock via RTCZero.  Not too many functions, but could break it into a separate clock module and then include those two as options.

May also change or consolidate date formatting class.

neilh10, 2019-03-15:
I wonder could it also support the PCF8523  - and perhaps be a generic rtc_sensor object defined at init - something like the adafruit/RTClib.
The SAMD has internal clock, however it doesn't survive  a power down - no Vrtc backupzone.   
Adafruit supply a clock board+SD with battery - https://www.adafruit.com/product/2922
It uses PCF8523  which has multiple options (including DS3231)
https://github.com/adafruit/RTClib

The PCF8523 is Digikey$0.93@100  +/-2secs/day
The DS3231 is Digikey $5.88@100  more accruate +/-2ppm or +/- 0.2Secs/day

SRGDamia1, 2019-03-16:
By this I mean to make a clock more generic, like a sensor is, so more could easily be added. But I'm not planning to actually add support for any more clocks right now. I don't have the time/need/hardware for anything else.