EnviroDIY/ModularSensors #246: setupXbee detail

neilh10, 2019-03-20:
Just checking if I've missed something in Xbee S6 WiFi setup - I think its minor but ....

in menu_a_lca_carte.ino
setupXbee() 
...        
#if defined(USE_XBEE_WIFI)
        tinyModem->sendAT(F("SO"),200);  // For WiFi - Disassociate from AP for Deep Sleep
"200" is not defined.

I think it should  be
        tinyModem->sendAT(F("SO"),100);  // For WiFi - Disassociate from AP for Deep Sleep
that is 0x40 is OFF to disassociate
0x100 is ON as manual says "Current implementation does not support this bit being turned off."
--

Using manual
XBee® Wi-Fi RF Module S6B 90002180 rev R June 2017
SO Command Pg 182 
https://www.digi.com/resources/documentation/digidocs/PDFs/90002180.pdf

SRGDamia1, 2019-03-20:
I'm pretty positive that's an error in the manual that was copied from the manuals for the radio bee's.  It doesn't even apply to pin sleep.

If you connect with XCTU you see this as the helper for sleep options: "Sleep Options Bit field. Bit 6 - Stay associated with AP during sleep. Bit 9 - Disassociate from AP for Deep Sleep. All other bits ignored."

SRGDamia1, 2019-03-20:
Honestly, I haven't tested the power draw difference between them.

closed 2019-03-20 by SRGDamia1

neilh10, 2019-03-21:
Hmm interesting thanks for the pointers