EnviroDIY/ModularSensors #247: Write up some info to help with maintaining dependencies

SRGDamia1, 2019-03-21:
For @fisherba and @neilh10

Tied to discussions in #213 and #235 

PlatformIO library dependency manager is awesome.. but not *quite* as awesome as conda.  You can end up with multiple copies of the same library installed.  Libraries won't be downgraded and conflicting dependencies don't get flagged.  If there are multiple versions installed, the builder will (I think) only pick one (the alphabetically first?) to build, but you could see weird errors or unexpected behavior.  **The fact that, per @aufdenkampe request, ModularSensors lists exact commits or version numbers of libraries as dependencies makes it much more likely that you'll end up with multiple copies of the same library.**  That being said, you shouldn't ever have to delete and re-download everything - that's just painful.

In short, if you need to update libraries in an already existing project:
- Open command line in project folder you're working in - whether it be your "develop" or "deployment" folder
    - Do this right in Atom or VSCode
- Update your libraries by typing ```pio lib update``` in the command line
- Run a "clean" in PlatformIO
    - either hit the button or type ```pio run --target clean``` into the command line
    - this will ensure all dependencies are installed at the right version
- Expand your .piolibdeps folder.  Look for any  libraries you have more than one of. 
- Delete all copies of any library you have more than one of. 
    - Either use the command ```pio lib uninstall xxx``` or just delete the offending folder
    - best to delete *all* copies of the offending repeat library
- Run a clean again
- Double check for repeats

fisherba, 2019-03-21:
Thanks for the instructions. I just did all of that, compiled, and still got the error message. Here's a screen shot of my results where you can see in the terminal: 
`Updating Adafruit ADS1X15                @ 7d67b45        [Up-to-date]`

And in the file structure that the ADS1X15 folder exists. 

And that I still get the compile error: 
`Error: Could not find `ADS1X15` dependency for `EnviroDIY_ModularSensors` library`

[image]

SRGDamia1, 2019-03-21:
Your screenshot is still showing repeats.  You must have two copies of ModularSensors in there.

I'll try myself again on VSCode.

SRGDamia1, 2019-03-21:
Worked fine for me on VSCode, too.

fisherba, 2019-03-21:
#facepalm. I was looking at a duplicate ini file... somehow in that same directory.

closed 2019-03-21 by fisherba

SRGDamia1, 2019-03-21:
Whew.. good.

aufdenkampe, 2019-04-02:
@SRGDamia1, thanks for that helpful opening comment https://github.com/EnviroDIY/ModularSensors/issues/247#issue-423925820 for this issue! It would be great to capture some of those instructions in our Wiki somewhere.

BTW, PlatformIO can do really well with managing multiple versions of libraries in the `.piolibdeps` folder. I do it all the time when I have different deployments using different versions. It *typically* does a great job of distinguishing different versions of the dependencies, using some hash magic.

That said, sometimes the libraries do get messed up, and things won't compile, so deleting stuff in the`.piolibdeps` folder is helpful. That's why we should preserve these great instructions for doing that efficiently.

w3asa, 2020-03-14:
How do I do this?
I have PlatformIO working with Atom and get errors with the blink sketch compile
Open command line in project folder you're working in - whether it be your "develop" or "deployment" folder
Do this right in Atom or VSCode

SRGDamia1, 2020-03-16:
@w3asa - this is now an out-dated issue.  The library definition for ModularSensors isn't referencing as many specific commits any more, the PlatformIO library manager has better handling for updates/duplicates, and the ```.piolibdeps``` structure is no longer used.