EnviroDIY/ModularSensors #250: Trigger an action using ModularSensors

acgold, 2019-04-14:
I'm specifically interested in triggering an ISCO automatic water sampler to collect water samples during storms, but the functionality could have many different applications.  I've seen an example of this functionality (open-storm.org: https://github.com/open-storm/perfect-cell/blob/master/perfect-cell.cydsn/autosampler.c), and I'm wondering if this might be possible within ModularSensors. If using the Mayfly, could it also need additional hardware like a relay board since an ISCO may be powered separately with a 12V battery?

neilh10, 2019-04-14:
Hi Adam
I think you could trigger the ISCO with a relay run from the TWI - eg Grove Relay @ 3V - you could look at the ISCO input, - if it wants a connection to gnd to trigger it, it may even be as simple as transistor 
I would think you want to define  what the trigger even is - I would think a rising hydrograph for the storm drain you are looking at  - so that would be  gage (SDI12) or even a simple float switch for when the water has got to a certain depth.  I haven't worked with ISCO but have heard a lot about them.

fisherba, 2019-04-23:
I just noticed this post. @aufdenkampe did this and has documentation here:
https://github.com/EnviroDIY/Trigger

He also did an external battery and monitored its voltage with the Mayfly (via another piece of hardware), but I am not sure where that is documented.

aufdenkampe, 2019-04-23:
@acgold, yes, we've done what you describe.

The key was to figure out exactly how to create a low-powered, solid-state latching contact closure switch, because the ISCO requires grounding a float line to "inhibit" the ISCO sampler program. All of that is documented here: https://github.com/EnviroDIY/Trigger/tree/master/hardware/ISCO-Mayfly-Trigger.

fisherba, 2019-04-23:
Here's a crude schematic of the final ISCO wiring as well as the voltage monitoring. The code for the trigger and external voltage can be found in the ["menu a la carte" example in ModularSensors](https://github.com/EnviroDIY/ModularSensors/tree/master/examples/menu_a_la_carte). 

[image]

acgold, 2019-04-23:
Thanks everyone for your help! This is great - exactly what I was l looking for.  Are timestamps recorded on the Mayfly when the trigger is activated, or are sample collection times saved on the ISCO (like how it runs when deployed on its own)?

aufdenkampe, 2019-04-23:
You're welcome! Sample collection times are saved on the ISCO.

closed 2019-04-23 by acgold