EnviroDIY/ModularSensors #253: -9999 for modem RSSI & SignalPercent in Sensor Testing Mode only

aufdenkampe, 2019-05-24:
@fisherba, @gcutrell (as described in https://github.com/EnviroDIY/ModularSensors/pull/252#issuecomment-495422031 & below) and now @brianjastram now all have observed this odd issue where in Sensor Testing mode, the modem is showing `-9999` for `Modem_RSSI` & `Modem_SignalPercent`, even when the data is successfully being sent to the Monitor My Watershed / EnviroDIY data sharing portal.

These are occurring with different radios and library versions.

@fisherba was using a GPRSBee rev.7b with the bleeding edge library at https://github.com/EnviroDIY/ModularSensors/pull/252.

@gcutrell was using a GPRSBee rev.6b with ModularSensors v0.21.4.

@brianjastram was using a Digi XBee3 LTE-M (with [EnviroDIY LTEbee-Adapter](https://github.com/EnviroDIY/LTEbee-Adapter)) with ModularSensors v0.21.4. Brian was getting 201 Server responses

@SRGDamia1, can we wrap this fix into #252?

SRGDamia1, 2019-05-24:
Oh, yes, I'm working on it now.

aufdenkampe, 2019-05-24:
Here's a bit more info.

When @brianjastram was using a Digi XBee3 LTE-M (with EnviroDIY LTEbee-Adapter) with ModularSensors v0.21.4, the data going to the SD card showed real values for for `Modem_RSSI` & `Modem_SignalPercent`.

However, when Jim was using a GPRSBee rev.6b with ModularSensors v0.21.4, the data going to the SD card did not show real values, even though they were logging to the server?!!

SRGDamia1, 2019-05-24:
Those are.. two unexpected things.  They acted oppositely?

aufdenkampe, 2019-05-24:
Not exactly oppositely.

Both logged to the Portal.

Both gave `-9999` values for modem variables in sensor testing mode.

The difference was that:
- the LTE-M gave real values when not in sensor testing mode (to SD card and in the REST POST JSON package
- the GPRSBee rev.6b gave `-9999` values for modem variables when not in sensor testing mode.

SRGDamia1, 2019-05-25:
Ok, that's more what I expected.  You can try again, but I'm still testing.

SRGDamia1, 2019-05-30:
I suspect this is all fixed in #252.  Can you check?

aufdenkampe, 2019-05-31:
@SRGDamia1, I just tested both Rev. 6b and Rev. 7b by compiling the head of develop, and both work without any need to change `modemStatusPin = 19` to `=-1`.  Awesome!

```ini
; Using ModularSensors *develop* branch (v0.22.2) as of 2019-05-30: head, 75 commits ahead of master (v0.21.4)
    https://github.com/EnviroDIY/ModularSensors.git#3b9142315a7801b3bc595ef10ec79d6041df4093
```

All your work for #252 definitely paid off! Thanks!

BTW, I tested on this sketch is at https://github.com/LimnoTech/EnviroDIY_deployments/tree/master/deploy-VLAWMO/VLAWMO-2.

closed 2019-05-31 by aufdenkampe