EnviroDIY/ModularSensors #255: GPRSBee R7 pin 12 doesn't indicate status by expected time

SRGDamia1, 2019-05-30:
Actually, not by expected time (2s) plus a nearly 2s of fudge.  Is it even wired up?

See conversations here:  #252

aufdenkampe, 2019-05-31:
For close-up photos of the two GPRSBee versions -- Rev. 6b & Rev. 7b -- see https://github.com/EnviroDIY/ModularSensors/pull/252#issuecomment-497545729.

fisherba, 2019-05-31:
So.. I went back to 
```const int8_t modemStatusPin = 19;```
on Rev7 modem at the status pin check commit:
```https://github.com/EnviroDIY/ModularSensors.git#3b9142315a7801b3bc595ef10ec79d6041df4093```
And modem status works. I think that's good news. I no longer need the pin on -1 for the Rev7 modem to work. 

In summary we have the following combinations all working successfully **after the status pin check commit** (all of my other tests were prior to that commit): 
Mayfly v5 + GPRSBeeR6b
Mayfly v5b + GPRSBeeR6b
Mayfly v5 + GPRSBeeR7
Mayfly v5b + GPRSBeeR7
(and it doesn't work with the LTE adapter, but it shouldn't as far as I understand)

fisherba, 2019-05-31:
Here's a debug with Mayfly v5b + GPRSBeeR7 combination with 
```const int8_t modemStatusPin = 19;```
on Rev7 modem at the status pin check commit:
```https://github.com/EnviroDIY/ModularSensors.git#3b9142315a7801b3bc595ef10ec79d6041df4093```

It works overall, but the first modem status didn't work, which was a replicatable result. 

[modem_log_baf_20190530d.txt](https://github.com/EnviroDIY/ModularSensors/files/3239642/modem_log_baf_20190530d.txt)