EnviroDIY/ModularSensors #257: Differential external voltage

brettmelbourne, 2019-06-06:
It looks like currently the external voltage sensor only supports single ended measurement. It would be great if differential measurement was also supported. Would that be hard to do? I'd be willing to take a stab. I have low mV range pyranomters [Apogee SP510](https://www.apogeeinstruments.com/sp-510-ss-upward-looking-thermopile-pyranometer/) 0-144mV and these are recommended for differential measurement.

SRGDamia1, 2019-06-07:
It shouldn't be hard at all. You'd could start with the external voltage cover, add a second data on, and can the ads library's read differential method instead of single ended in the add single measurement really code. You may also need to adjust the gain.

neilh10, 2019-06-07:
I agree. One thing I believe the part is ADS1115 (not what is on the mayfly schematic).

You would of course need to connect the Apogee SP510 gnd wire to the digital Mayfly gnd, and then my reading of the datasheet one data wire to AIN3 (or I think AIN1)  and the other data wire to the AIN0 or 2 and specify it in MUX[]

The data sheet says the gain can be set to +-256mV -  (which is 512mV for 16bits) so you should be able to get a resolution 512mV/2^^16  or  7.8uV

Then in the ADS11x data sheet it says 
10.1.2 Single-Ended Inputs
.....
The ADS1115 also allows AIN3 to serve as a common point for measurements by appropriate setting of the MUX[2:0] bits. AIN0, AIN1, and AIN2 can all be measured with respect to AIN3. In this configuration, the ADS1115 operates with inputs, where AIN3 serves as the common point. This ability improves the usable range over the single-ended configuration because negative differential voltages are allowed when GND < V (AIN3) < VDD; however, common-mode noise attenuation is not offered.

brettmelbourne, 2019-06-10:
Thanks to you both for these comments. I have the SP510 working properly on differential measurement with some basic arduino code. This is using the +-256mV range. I'll have a go at a modularsensors version for differential voltage.

neilh10, 2020-07-13:
BTW - the INA219 is effectively a single differential sensor, with a common mode range to +26V.

mbarneytu, 2021-01-21:
Has anyone implemented differential signaling capability for the ExternalVoltage sensor? I'm currently pursuing it myself. Since ExternalVoltage is built specifically for the TI ADS1015 and 1115 chips, it would be nice to enable more of their underlying functionality. Here are the items I have in mind:
- differential signaling
- enable the full 5V analog input capability of the chip. ExternalVoltage currently rejects any reading > 3.6V
- fully implement the Programmable Gain Amplifier (PGA) of the chip. Right now, the gain value is always set to 1 on the ADC (`ads.setGain(GAIN_ONE);`), and the digital output from the ADC is just multiplied by the "gain" value that the user supplies when instantiating ExternalVoltage.

It looks like all of these are supported by the underlying fork of the Adafruit_ADS1015 library, so I think it should be doable, and would broaden the capabilities of MS for additional analog sensors. I'm just trying to pursue implementation in ways that don't break backward compatibility (and my C++ skills are a bit rusty). I'm interested in everyone's input.

-Matt

mbarneytu, 2021-01-21:
As additional context, I'm using the ADS1115 to read a 0-5V signal, so am powering it via the Mayfly's 5V boost regulator. In order to control the ADS over I2C, I'm using an external level-shifter so that it's compatible with Mayfly's 3V3 I2C pins.

neilh10, 2021-01-21:
As I understand the ADS1115, the inputs can't exceed the VDD+0.3V in 7.1 Absolute Max Rating 
and there are a few other references to not exceeding the VDD+0.3V
Its powered off Vcc/3.3V.  So that is probably where your 3.6V come from if using the Mayfly ADS1115 . 
However IHMO Vcc=3.3V is not a safe bet, due to the MAYFLY power design, and I use a lower abs max.
Edges at 0V or the top end are always a challenge. 

The reason is its  depending on how the LiIon battery power is used and number of active sensors..  
All the sensors are turned on together. 
Probably 3.3V or even 3.0V max input for accurate top end measurement.
The 3.3V is provided by SPX3819, an LDO that can have a dropout of 700mV if the regulator is fully loaded at 500mA (unlikely). 
If you can guarantee the power draw is less than 150mA when you are taking your measurements, the max drop out is 450mV.
So if your battery voltage is above 3.75V  (3.3+0.45) you can probably rely on Vcc=3.3V being accurate
For a basic load of 50mA, max drop out is 80mV - that is LiIon battery down to 3.4V providing an accurate Vcc=3.3V.

For level shifting, part of the issue becomes what accuracy do you need. 
for 0-5V, you have a level shift required,  which is two R - and its cheap enough to usually specify 0.1%. For two Rs, the inaccuracy becomes 0.2%  0.1% is roughly equivalent to 10bit ADC.
So if one pin is always likely to be in the 0-Vcc range, it won't need a level shifter. If it can't be guaranteed, then you need another two R's and the error becomes 0.4% absolute.
Now add the ERROR  to the ADC sampling quantization,  for the range specified and you have an idea of the accuracy that you can measure to of your analog signal.
If the sensor signal could be ratio metric with the source of the ADC - then further error calculations could be zeroed out.
The ADS1115 has its own internal Vref for the ADC - which is mostly very nice, but in this case you have an externally generated +5V  ~ so hopefully your sensor has an absolute value output. Usually a differential signal gets round what the the Vcc is.

Another way is to move the ADS1115 offboard where your level shifter would be, and power it off the Boost 5V, which is pretty well guaranteed.

For a higher accuracy differential "high voltage" measurement,  I'm using the INA219  but that is only one ADC, but all the potential sources of errors are calculated for you.

mbarneytu, 2021-01-21:
Hi Neil,

> Another way is to move the ADS1115 offboard where your level shifter would be, and power it off the Boost 5V, which is pretty well guaranteed.

Right, this is what I've done; sorry that wasn't clear.