EnviroDIY/ModularSensors #259: New Zealand time zone doesn't work

escaped-echidna, 2019-06-17:
I have tried to run the ModularSensors 'No Cellular' and 'simple logging' script with a time zone of 12 (Auckland). This gives me a time stamp with an offset of 20 hours, not 12 hours, for some reason.

SRGDamia1, 2019-06-17:
Do you have your clock set to be in UTC or in local time? All the examples assume your clock is on UTC. If you'd prefer the clock to be in another time zone, you have to change lines 161-165 accordingly.

I'll also test on my end to see what I get.

escaped-echidna, 2019-06-24:
When I run the PCsync program to set the RTC clock, it correctly sets to UTC time, regardless of the time I have set on my computer. However when I then run the 'simple logging' script it gives the wrong time. Using time zones between -9 and +9 it works, just not for time zones -10, -11, -12 or +10, +11, +12.
[image]
[image]

SRGDamia1, 2019-06-24:
I'm sorry, I haven't gotten a chance to look into this yet.  I will try to get to it this week.

SRGDamia1, 2019-06-24:
Found the issue!  I wasn't explicitly casting the timezone to a long into before multiplying by 3600 to get from hours to seconds, so once the timezone was bigger than 9 (9.1, actually) it was overflowing and not working.

That PCSync program will always set the clock in UTC.  The sync inside of modular sensors will set the clock to whatever time zone you want (now, even correctly!).

Thank you so much for posting an issue from the other side of the world!  I probably never would have found the problem otherwise.

commit 9ce918f2 (2019-06-24): Explicit cast to uint32_t before multiplication, fixes #259.  Version bump
closed 2019-06-24 by SRGDamia1