EnviroDIY/ModularSensors #272: EnviroDIYPOST.begin()

neilh10, 2019-08-02:
For the lastest release 0.23.2 ( some great refactoring ...)
just wondering how EnviroDIYPublisher is supposed to start.

I'm modeled after menu_a_la_carte.ino - and to get it posting I needed to add this equivalent to setup():

    EnviroDIYPOST.begin(dataLogger, ....

so it looks like this in setup()
    // Begin the logger
    dataLogger.begin();
    EnviroDIYPOST.begin(dataLogger, &modemPhy.gsmClient,registration_token,provider.s.sampling_feature);    

However maybe I'm missing something, as I know it used to work in the previous release with out 
 EnviroDIYPOST.begin(..)  I just can't figure out why.

SRGDamia1, 2019-08-06:
You shouldn't need that begin, unless you used an empty constructor for the EnviroDIYPOST object.

There are two types of "begin" functions in the Arduino world:
- There are "begin" functions that setup hardware parameters and pin modes - these are always required because they are things that physically need to happen in the hardware to make things work.  In this library, I've tried to label this type of "begin" function _setup()_, but the LoggerBase is a pretty gaping hole in that convention.
- There are "begin" functions that merely set variable values within a class objects.  These are essentially overloads or replacements for adding arguments to the object constructor.  The only reason these functions exist is to make absolutely certain that the compiler creates objects in the right order.  Because Arduino hides the true ```main()``` function, it's theoretically possible that global variables you construct outside of the ```setup()``` or ```loop()``` functions would be seen in the wrong order by the compiler and you might end up attempting to reference an object that hasn't yet been assigned a place in memory in the constructor for another object.  Offloading all assignment of internal values for a class from the constructor into a begin called in the ```setup()``` makes 100% certain that will not happen because it's all assigned at runtime.  In reality, I've never had the slightest problem with complicated constructors and would rather use them.  I implemented the overload-style ```begin()``` for some of the classes a while back because I was afraid the compiler might have been to blame for some issues I was having.  It never was the problem, but I left the functions in anyway.

neilh10, 2019-08-06:
I totally agree with the .begin() and ensuring the order of init.  I totally design like that.   

So I'm closing this as assuming that   menu_a_la_carte.ino  works for you.

For me I can't see how the constructor started the dataPublishing::  that is in the begin() - so raised this as a flag.

 I am using     (ie not empty and three constructors are similar)    

EnviroDIYPublisher EnviroDIYPOST(dataLogger, 15,0);

but still can't see how  say this constructor invokes the dataPublishing::  that is in .begin() 
```
EnviroDIYPublisher EnviroDIYPOST(dataLogger, &modem.gsmClient, registrationToken, samplingFeature);

EnviroDIYPublisher::EnviroDIYPublisher(Logger& baseLogger, Client *inClient,
                                 const char *registrationToken,
                                 const char *samplingFeatureUUID,
                                 uint8_t sendEveryX, uint8_t sendOffset)
  : dataPublisher(baseLogger, inClient, sendEveryX, sendOffset)
{
    setToken(registrationToken);
    _baseLogger->setSamplingFeatureUUID(samplingFeatureUUID);
    // MS_DBG(F("dataPublisher object created"));
}
```

closed 2019-08-06 by neilh10

SRGDamia1, 2019-08-07:
The EnviroDIYPublisher constructor invokes the dataPublisher constuctor (with arguments pass in the initializer list).  It's the dataPublisher constuctor  that calls ```_baseLogger->registerDataPublisher(this);```.