EnviroDIY/ModularSensors #276: Develop wrapper for Meter Teros 11 Soil Moisture Sensors

aufdenkampe, 2019-08-14:
The Meter Group purchased Decagon about a year ago, and just recently stopped selling the 5TM soil moisture sensor, replacing it with the [Teros 11 Soil Moisture + Temperature](https://www.metergroup.com/environment/products/teros-11/) sensor, which uses a more robust construction design that Decagon had been piloting.

Unfortunately @Volk3rJ discovered that the SDI-12 commands are no longer the same. These differences include:
1. A change in the sensor response, where the Teros 11 gives raw counts rather than EA values as 1st of 2 returned variables. See https://github.com/LimnoTech/EnviroDIY_deployments/commit/23a6f688ae6edadf3a5e91030296bd79364fc9c4
2.  A change in the SDI-12 timing, which gives lots of errors when more than one Teros 11 sensor is attached to the same Mayfly.

@Volk3rJ partially solved #2 by placing the sensor-variables as far apart as possible in the variables array. He also noticed some hints of these differences in the [TEROS 11/12 INTEGRATOR GUIDE](http://publications.metergroup.com/Integrator%20Guide/18224%20TEROS%2011-12%20Integrator%20Guide.pdf) and the [TEROS 11/12 USER MANUAL](http://publications.metergroup.com/Manuals/20587_TEROS11-12_Manual_Web.pdf).

Volk3rJ, 2019-08-19:
Regarding the SDI-12 timing issue: 

The sensor that gets called first responds normal, the sensor called second only reads error values (or maybe doesn't respond at all?). I partially fixed this by calling modem RSSI and other values in the array between 1. and 2. sensor. This curiously works for one of my stations but not for another! 

I also tried to increase the sensor delays and timings in the 5TM.h file without any luck

aufdenkampe, 2021-03-22:
Note that the timing issues here popped up again for @shepardb when he updated the ModularSensors library on a station deployed with 2 Teros 11 sensors that had been working fine for a year.

aufdenkampe, 2021-05-05:
Reopening and looping @GArrigotti-cws into this issue, so we can fix this issue that has returned since Modular Sensors versions >0.25ish (@shepardb noticed it summer of 2020).

SRGDamia1, 2021-05-05:
The changes I just made with v0.28.4 *might* help this.  I made the time between the SDI-12 break and the actual command settable per sensor.  I discovered with the newest Meter sensor we got, it was not responding properly with even the 10ms delay I'd added for the RDO.  I set the delay for the Teros (and 5TM) back to 0, which is what it was back in 2019.  If it doesn't fix the problem, it at least gives one other aspect of the timing that you can try to tweak to see what works.

For any sensor that reports back a manufacturer of "METER" I also added code to turn off the DDI serial output string.  If the sensor isn't at the factory default address of 0, it shouldn't be putting out the string anyway, but it seem safer to turn it off just in case it was causing extra garbage on the SDI-12 line.

GArrigotti-cws, 2021-05-05:
@SRGDamia1 I did tweak a couple of those and still had some issues, I'll see if that change in modular sensors partially rectifies.

SRGDamia1, 2021-05-05:
@GArrigotti-cws - it's a bit of a long-shot, but maybe it changes something.

In the 'tools' section of the SDI12 library there are a few scripts that I made to test sensor timing - basically repeating everything over and over with slight changes to the warm-up, wake delay, etc.  They might help.

GArrigotti-cws, 2021-05-05:
@SRGDamia1 Okay, I'll give it a try.  Thank you.

aufdenkampe, 2021-05-05:
@SRGDamia1, thank you so much for describing these issues and making the new release!

As @GArrigotti-cws, mentioned, a couple of weeks ago the two of us modified the Teros 11 timing settings in `src/sensors/MeterTeros11.h`:

```cpp
/// @brief Sensor::_warmUpTime_ms; the Teros 11 warm-up time in SDI-12 mode:
/// 245ms typical
#define TEROS11_WARM_UP_TIME_MS 250
/// @brief Sensor::_stabilizationTime_ms; the Teros 11 is stable after 50ms.
#define TEROS11_STABILIZATION_TIME_MS 50
/// @brief Sensor::_measurementTime_ms; the Teros 11 takes25 ms to 50 ms to
/// complete a measurement.
#define TEROS11_MEASUREMENT_TIME_MS 50
```

I think we increased the warm time to seconds, but it didn't make a difference.

If I understand correctly, with this commit https://github.com/EnviroDIY/ModularSensors/commit/a6eee7595b4b3cbe55df761818ac3759dda5d137#diff-c2718078e9b993d65131fd0cc535a573e7e2d2130591c9a77021d3f7519ae383 \
you added this:

```cpp
/// @brief Extra wake time required for an SDI-12 sensor between the "break"
/// and the time the command is sent.  The Terros-11 requires no extra time.
#define TEROS11_EXTRA_WAKE_TIME_MS 0
```

and that we might want to try increasing it from 0 to 10 ms or maybe higher. Is that right?

aufdenkampe, 2021-05-05:
@GArrigotti-cws, the other thing we'll try is to power the Teros 11 with 5V. I just learned from @SRGDamia1 that the Mayfly's ATmega1284p microprocessor can actually handle up to 6V.

To protect the Mayfly, we've always powered Meter sensors at 3.3V to match the Mayfly's voltage. Although most Meter sensors say they need a minimum power of 3.6 V to 4.0 V, we have found that they operate fine at 3.3. However, maybe the Teros 11 is more sensitive at low voltage. Also, it's stated minimum is 4.0 V, unlike many of the others that list 3.6V. 

See http://publications.metergroup.com/Integrator%20Guide/18224%20TEROS%2011-12%20Integrator%20Guide.pdf

SRGDamia1, 2021-05-06:
Yes, you could try increasing the extra wake time. For a long time it was zero, then I bumped it up to 10ms for the RDO. I found the newest Meter sensors *did not* respond consistently with the delay so I made it an argument so it could be set back to 0 for the Meter sensors.

The extra wire time applies to every SDI12 command sent to the device. Since the whole SDI12 is designed for environmental sensors; the sensors are required to go to sleep any time except when they are being addressed or measuring. The sensors are required to be ready to listen for a new command within 100 ms of a "break" and "marking" on the SDI12 line. The Meter sensors don't need any of that 100ms, they're ready within the space of the break and marking. The newest CTD we got actually was a little flaky when given the extra time. I think some of the Campbell sensors need the full 100ms.

So, for a Meter sensor, I suspect the right amount of time is 0, but you can try increasing it. It should never be over 100.

Also, the maximum voltage on a pin for the 1284p is dependent on the voltage it's powered at; it should be no more than 0.5V higher than the power voltage. The Mayfly powers it with 3.3V, so the pin max is 3.8V. It's as much voltage differential as absolute voltage. The very maximum power voltage it can handle is 6V, but that's not what the Mayfly has. So, while I really don't think short pulses like SDI12 at over voltage are really likely to be problematic, I definitely would not recommend regularly subjecting the Mayfly pins to 5V for long periods of time.

aufdenkampe, 2021-05-06:
@SRGDamia1, that's a very helpful explanation! Thanks.

I just looked again at the [Meter Teros 11-12 Integrators Guide](http://publications.metergroup.com/Integrator%20Guide/18224%20TEROS%2011-12%20Integrator%20Guide.pdf), and it says that output voltage is typically 3.6V!  

I had thought the SDI-12 specification was 5V for SDI-12 sensors, and presumed the Teros 11 followed that.

So for the Meter Teros 11 or 12 sensors, even if they are powered at ~12V, their digital high signals should still be ~3.6V. 

The [Meter Hydros 21/CTD Integrators Guide] (http://library.metergroup.com/Integrator%20Guide/18281_HYDROS21(CTD)_GEN1.pdf) shows the digital output voltage of ~ 3.6V.

So, it seems we should be powering these Meter sensors at 5V.

neilh10, 2021-05-06:
Hello I'm wondering where the info on the mega1284 handling up to 6V (on a input pin from SDI-12?) came from.
I read from the data sheet I read for the Mayfly it would be a max of 3.8V?

In the data sheet I've been using from the BOM ATMEGA1284P_VQFN44 or ATMEGA1284P-MU it says the Absolute maximum ratings, Voltage on any pin except RESET is to Vcc+0.5V.
Mayfly Vcc is 3.3V, so that would mean a max of 3.8V on any input pin.
It does say that the Maximum operating voltage is 6.0V, Vcc can be 1.8Vto5.5V, and DC Current per I/O pin 40mA - though test conditions for 3V is a sink of 10mA

[image]
https://ww1.microchip.com/downloads/en/DeviceDoc/ATmega164A_PA-324A_PA-644A_PA-1284_P_Data-Sheet-40002070B.pdf

neilh10, 2021-05-06:
Reading the  Meter Teros 11 spec, it does fit with what SDI-12 spec.  
I would say for reliability and repeatability you want to power it with 4V or more per the Meter Teros 11 spec. It may work at a lower voltage in the lab, but might not across all operating voltages of -40C to +60C.
 The SDI-12specs says  the data-line  should exceed 3.5V ... but may be up to 5.5V  - that fits with the Teros 11 Typical 3.6V.  Teros 11 would still be in spec to supply 5.5V (maybe this only happens on a blue-moon at -10C, but would still be in specification - but might cause problems for the Mayfly ) or maybe it never happens due to circuit design. 
My understanding of crimping the specifications, is that you may get something working in nice conditions indoors, but what happens in the great outdoors.
My tradeoff is to limit the current into the Mayfly with a series R ~ not totally correct, but seems like its closer to the requirements of what is needed. That is also the SDI-12 recommended surge protection
[image]

aufdenkampe, 2021-05-06:
@GArrigotti-cws and I are testing right now. Seems like 5V help a bit. We now have both sensors working in Sensor Testing Mode, but only one works in regular logging mode.

aufdenkampe, 2021-05-06:
Update from testing this afternoon:
- Sensor Testing Mode is buggy, and hasn't been working since we started playing with turning on debugging.
- When we Define `MS_VARIABLEARRAY_DEBUG`, suddenly we got both sensors to work. Only one sensor at a time was reporting when it was not defined.
  - NOTE that `MS_SDI12SENSORS_DEBUG` showed that both sensors were responding. However, even with `TEROS11_WARM_UP_TIME_MS 10000`, the first sensor in the list would not ever respond on the first try.

Here's the example of it working with both debug statements defined:

```
> Executing task: C:\Users\ArrigottiG\.platformio\penv\Scripts\platformio.exe device monitor <
 
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM4  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
 
 
Now running Arrigotti_NoRadio.ino on Logger Arrigotti
 
Using ModularSensors Library version 0.28.4
Logger timezone is set to UTC -8
RTC timezone is set to UTC
Current RTC time is: 2000-05-19T07:09:32-08:00
All variable UUID's appear to be correctly formed.
 
2a5f077b-8de9-45ee-9704-4743726101fb -> SoilEa
168081de-2c1a-458f-8f86-1d88f7b8c394 -> SoilTemp
d7a3e8c8-d1f0-4503-a76f-93f5b6861850 -> SoilVWC
e5d304df-995b-4272-9a26-f3b525c315b7 -> BoschBME280Temp       
f8984e98-bbd4-4c29-a178-a64bcd546411 -> BoschBME280Humidity   
6872a270-5dfb-4cee-9326-118adc97af9e -> BoschBME280Pressure   
c3a10820-59ef-48a0-9fa5-88bb06914186 -> Battery
61e38606-b91c-4058-ac8c-3ff1ff280ede -> BoardTemp
29abed47-770f-44ef-893e-bff2645a1ff9 -> SoilEa
aa142833-d52e-4f91-b2b7-d3fa953acdae -> SoilTemp
49d31d59-1986-4a9c-ab6f-16d7639c8abd -> SoilVWC
 
There are 0 calculated variables in the group. <--VariableArray
There are 0 calculated variables in the group. <--VariableArray
This logger has a variable array with 11 variables, of which 11 come from 5 sensors and 0 are calculated.
Logger portion of setup finished.
 
Setting up sensors...
Beginning setup for sensors and variables... <--VariableArray
Running sensor setup functions. <--VariableArray
    EnviroDIY Mayfly at EnviroDIY Mayfly was already set up! <--VariableArray
    Set up of MeterTeros11 at SDI12-1_Pin7 ... <--VariableArray
Enabling interrupts for SDI12 on pin 7 <--SDI12Sensors
  Asking for sensor acknowlegement <--SDI12Sensors   
    MeterTeros11 at SDI12-1_Pin7 did not reply! <--SDI12Sensors
    MeterTeros11 at SDI12-1_Pin7 replied as expected. <--SDI12Sensors
  Getting sensor info <--SDI12Sensors
   SDI12 Address: 1 <--SDI12Sensors   
   SDI12 Version: 1.30 <--SDI12Sensors
   Sensor Vendor: METER <--SDI12Sensors
   Sensor Model: TER11 <--SDI12Sensors
    Sensor Version: 112 <--SDI12Sensors
   Sensor Serial Number: T11-00006560 <--SDI12Sensors
  Suppressing DDI string on Meter sensor <--SDI12Sensors
        ... setup succeeded. <--VariableArray
    Set up of BoschBME280 at I2C_0x76 ... <--VariableArray
        ... setup succeeded. <--VariableArray
    Set up of MaximDS3231 at I2C_0x68 ... <--VariableArray
        ... setup succeeded. <--VariableArray
    Set up of MeterTeros11 at SDI12-6_Pin7 ... <--VariableArray
Enabling interrupts for SDI12 on pin 7 <--SDI12Sensors
  Asking for sensor acknowlegement <--SDI12Sensors
    MeterTeros11 at SDI12-6_Pin7 replied as expected. <--SDI12Sensors
  Getting sensor info <--SDI12Sensors
   SDI12 Address: 6 <--SDI12Sensors
   SDI12 Version: 1.30 <--SDI12Sensors
[... 152 lines cut ...]
*****--- 4 sensors now complete ---***** <--VariableArray
10 . 3 --->> Collected result of reading 3 from MeterTeros11 at SDI12-6_Pin7 ... <--VariableArray
   ... got measurement result. <<--- 10 . 3 <--VariableArray
10 --->> Finished all measurements from MeterTeros11 at SDI12-6_Pin7 , putting it to sleep. ... <--VariableArray
   ... succeeded in putting sensor to sleep. <<--- 10 <--VariableArray
2 --->> MeterTeros11 at SDI12-1_Pin7 powered down. <<--- 2 <--VariableArray
5 --->> BoschBME280 at I2C_0x76 powered down. <<--- 5 <--VariableArray
10 --->> MeterTeros11 at SDI12-6_Pin7 powered down. <<--- 10 <--VariableArray
*****--- 5 sensors now complete ---***** <--VariableArray
----->> Averaging results and notifying all variables. ... <--VariableArray
--- Averaging results from MeterTeros11 at SDI12-1_Pin7 --- <--VariableArray
--- Notifying variables from MeterTeros11 at SDI12-1_Pin7 --- <--VariableArray
--- Averaging results from BoschBME280 at I2C_0x76 --- <--VariableArray
--- Notifying variables from BoschBME280 at I2C_0x76 --- <--VariableArray
--- Averaging results from EnviroDIY Mayfly at EnviroDIY Mayfly --- <--VariableArray
--- Notifying variables from EnviroDIY Mayfly at EnviroDIY Mayfly --- <--VariableArray
--- Averaging results from MaximDS3231 at I2C_0x68 --- <--VariableArray
--- Notifying variables from MaximDS3231 at I2C_0x68 --- <--VariableArray
--- Averaging results from MeterTeros11 at SDI12-6_Pin7 --- <--VariableArray
--- Notifying variables from MeterTeros11 at SDI12-6_Pin7 --- <--VariableArray
... Complete. <<----- <--VariableArray
 
\/---- Line Saved to SD Card ----\/
2000-05-19 07:10:00,1.26399,22.10,0.000,21.28,43.707,101220.54,4.852,24.00,1.26615,21.60,0.000
```

GArrigotti-cws, 2021-05-06:
Oddly enough, in if you put it into sensor testing mode now both sensors will work.  We increased our wake time to five to ten seconds, but it will fail in normal logging interval.  But will will allow the normal interval to work if you enable the variable array debug in the .ini file.  @aufdenkampe  will post my output-  Still really odd.

aufdenkampe, 2021-05-18:
@SRGDamia1, we've found a bug where we can only read 1 of 2 Teros11 sensors unless we have defined `MS_VARIABLEARRAY_DEBUG`. As soon as we do, both sensors read fine in the same sketch. When we don't define `MS_VARIABLEARRAY_DEBUG`, both sensors work on their own when the other is unplugged. 

Any ideas on what might be causing this bug?

I plan on working on this tomorrow (thanks @GArrigotti-cws for sending me two Teros11 sensors), so I could do specific testing.

SRGDamia1, 2021-05-24:
No, scanning all the debugging prints, I don't see anything in the variable array debug that would lead to that.  It's particularly odd because if it was an error there, I'd guess we'd have seen it much earlier.

Do you have your SDI-12 addresses set one after the other or are they spaced out?  I find sometimes the sensors get confused if I use "1" and "2" but putting space between them (ie "1" and "4") helps.  I suspect that has something to do with the not-quite-exactly-perfect timing in the SDI-12 library due to bit-banging and interrupt hand-offs.

Could you please post turn on debugging for SDI-12 and the Terros and then post the full log from a couple of readings with and without the variable array debug?  Obviously the output from the latter will be much longer than the former, but I should be able to line them up and see more of what's going on.

ie, first use and post output from :
```
-DMS_SDI12SENSORS_DEBUG
-DMS_METERTEROS11_DEBUG
```

then use and post output from:
```
-DMS_SDI12SENSORS_DEBUG
-DMS_METERTEROS11_DEBUG
-DMS_VARIABLEARRAY_DEBUG
```

GArrigotti-cws, 2021-05-24:
@SRGDamia1  Here should be the information, I have the address one and six.  I also placed sensor one at the top of the variable array and moved the other one to the bottom.  @aufdenkampe and I did play with some of the timing issues also in varying configurations.

const char*   teros11SDI12address = "1"; 
const int8_t  terosPower = sensorPowerPin; 
const int8_t  terosData  = 7;              
const uint8_t teros11NumberReadings = 3;  

MeterTeros11 teros11(*teros11SDI12address, terosPower, terosData, teros11NumberReadings);

const char*   teros11_2SDI12address = "6"; 
MeterTeros11 teros11_2(*teros11_2SDI12address, terosPower, terosData, teros11NumberReadings);

------------------------------------
> Executing task: C:\Users\ArrigottiG\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM4  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

Now running Arrigotti_NoRadio.ino on Logger Arrigotti

Using ModularSensors Library version 0.28.4
Logger timezone is set to UTC -8
RTC timezone is set to UTC
Current RTC time is: 2000-06-06T03:48:12-08:00
All variable UUID's appear to be correctly formed.

2a5f077b-8de9-45ee-9704-4743726101fb -> SoilEa
168081de-2c1a-458f-8f86-1d88f7b8c394 -> SoilTemp
d7a3e8c8-d1f0-4503-a76f-93f5b6861850 -> SoilVWC
e5d304df-995b-4272-9a26-f3b525c315b7 -> BoschBME280Temp    
f8984e98-bbd4-4c29-a178-a64bcd546411 -> BoschBME280Humidity
6872a270-5dfb-4cee-9326-118adc97af9e -> BoschBME280Pressure
c3a10820-59ef-48a0-9fa5-88bb06914186 -> Battery
61e38606-b91c-4058-ac8c-3ff1ff280ede -> BoardTemp
29abed47-770f-44ef-893e-bff2645a1ff9 -> SoilEa
aa142833-d52e-4f91-b2b7-d3fa953acdae -> SoilTemp
49d31d59-1986-4a9c-ab6f-16d7639c8abd -> SoilVWC

This logger has a variable array with 11 variables, of which 11 come from 5 sensors and 0 are calculated.
Logger portion of setup finished.

Setting up sensors...
Enabling interrupts for SDI12 on pin 7 <--SDI12Sensors
  Asking for sensor acknowlegement <--SDI12Sensors
    MeterTeros11 at SDI12-1_Pin7 did not reply! <--SDI12Sensors
    MeterTeros11 at SDI12-1_Pin7 replied as expected. <--SDI12Sensors
  Getting sensor info <--SDI12Sensors
   SDI12 Address: 1 <--SDI12Sensors
   SDI12 Version: 1.30 <--SDI12Sensors
   Sensor Vendor: METER <--SDI12Sensors
   Sensor Model: TER11 <--SDI12Sensors
    Sensor Version: 112 <--SDI12Sensors
   Sensor Serial Number: T11-00006560 <--SDI12Sensors
  Suppressing DDI string on Meter sensor <--SDI12Sensors
Enabling interrupts for SDI12 on pin 7 <--SDI12Sensors
  Asking for sensor acknowlegement <--SDI12Sensors
    MeterTeros11 at SDI12-6_Pin7 replied as expected. <--SDI12Sensors
  Getting sensor info <--SDI12Sensors
   SDI12 Address: 6 <--SDI12Sensors
   SDI12 Version: 1.30 <--SDI12Sensors
   Sensor Vendor: METER <--SDI12Sensors
   Sensor Model: TER11 <--SDI12Sensors
    Sensor Version: 112 <--SDI12Sensors
   Sensor Serial Number: T11-00006574 <--SDI12Sensors
  Suppressing DDI string on Meter sensor <--SDI12Sensors
Setting up file on SD card
Data will be saved as Arrigotti_2000-06-06.csv
Putting processor to sleep
[... 325 lines cut ...]
2 results expected This differs from the sensor's standard design of 3 measurements!!
    Concurrent measurement started. <--SDI12Sensors
  Asking for sensor acknowlegement <--SDI12Sensors
    MeterTeros11 at SDI12-1_Pin7 replied as expected. <--SDI12Sensors
  Beginning concurrent measurement on MeterTeros11 at SDI12-1_Pin7 <--SDI12Sensors
  Beginning concurrent measurement on MeterTeros11 at SDI12-1_Pin7 <--SDI12Sensors
  Beginning concurrent measurement on MeterTeros11 at SDI12-1_Pin7 <--SDI12Sensors
  Beginning concurrent measurement on MeterTeros11 at SDI12-1_Pin7 <--SDI12Sensors
  Beginning concurrent measurement on MeterTeros11 at SDI12-1_Pin7 <--SDI12Sensors
2 results expected This differs from the sensor's standard design of 3 measurements!!
    Concurrent measurement started. <--SDI12Sensors
  Asking for sensor acknowlegement <--SDI12Sensors
    MeterTeros11 at SDI12-6_Pin7 replied as expected. <--SDI12Sensors
  Beginning concurrent measurement on MeterTeros11 at SDI12-6_Pin7 <--SDI12Sensors
  Beginning concurrent measurement on MeterTeros11 at SDI12-6_Pin7 <--SDI12Sensors
  Beginning concurrent measurement on MeterTeros11 at SDI12-6_Pin7 <--SDI12Sensors
  Beginning concurrent measurement on MeterTeros11 at SDI12-6_Pin7 <--SDI12Sensors
  Beginning concurrent measurement on MeterTeros11 at SDI12-6_Pin7 <--SDI12Sensors
2 results expected This differs from the sensor's standard design of 3 measurements!!
    Concurrent measurement started. <--SDI12Sensors

 \/---- Line Saved to SD Card ----\/
2000-06-06 03:53:00,1.22668,21.80,0.000,23.25,42.572,101447.06,4.761,26.50,-9999.00000,-9999.00,0.000

------------------------------------------

SRGDamia1, 2021-05-24:
@GArrigotti-cws Could your turn on SDI-12 deep and Terrow (`-DMS_SDI12SENSORS_DEBUG_DEEP
-DMS_METERTEROS11_DEBUG`) and post _two different logs_, running exactly the same program, one with `-DMS_VARIABLEARRAY_DEBUG` and one without?

GArrigotti-cws, 2021-05-24:
@SRGDamia1 -DMS_SDI12SENSORS_DEBUG_DEEP fails to build.

GArrigotti-cws, 2021-05-24:
@SRGDamia1 with Meter Teros:
> Executing task: C:\Users\ArrigottiG\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM4  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

Now running Arrigotti_NoRadio.ino on Logger Arrigotti

Using ModularSensors Library version 0.28.4
Logger timezone is set to UTC -8
RTC timezone is set to UTC      
Current RTC time is: 2000-06-06T04:26:02-08:00
All variable UUID's appear to be correctly formed.

2a5f077b-8de9-45ee-9704-4743726101fb -> SoilEa
168081de-2c1a-458f-8f86-1d88f7b8c394 -> SoilTemp
d7a3e8c8-d1f0-4503-a76f-93f5b6861850 -> SoilVWC
e5d304df-995b-4272-9a26-f3b525c315b7 -> BoschBME280Temp    
f8984e98-bbd4-4c29-a178-a64bcd546411 -> BoschBME280Humidity
6872a270-5dfb-4cee-9326-118adc97af9e -> BoschBME280Pressure
c3a10820-59ef-48a0-9fa5-88bb06914186 -> Battery
61e38606-b91c-4058-ac8c-3ff1ff280ede -> BoardTemp
29abed47-770f-44ef-893e-bff2645a1ff9 -> SoilEa
aa142833-d52e-4f91-b2b7-d3fa953acdae -> SoilTemp
49d31d59-1986-4a9c-ab6f-16d7639c8abd -> SoilVWC

This logger has a variable array with 11 variables, of which 11 come from 5 sensors and 0 are calculated.
Logger portion of setup finished.

Setting up sensors...
Setting up file on SD card
Data will be saved as Arrigotti_2000-06-06.csv
Putting processor to sleep

----- WARNING ----- !!!!!!!!!!!!!!!!!!!!
!!!!!!!!!! ----- WARNING ----- !!!!!!!!!!
!!!!!!!!!!!!!!!!!!!! ----- WARNING ----- 
 
The current clock timestamp is not valid!
 
----- WARNING ----- !!!!!!!!!!!!!!!!!!!!
!!!!!!!!!! ----- WARNING ----- !!!!!!!!!!
!!!!!!!!!!!!!!!!!!!! ----- WARNING ----- 
------------------------------------------
2 results expected This differs from the sensor's standard design of 3 measurements!!
2 results expected This differs from the sensor's standard design of 3 measurements!!
MeterTeros11 at SDI12-1_Pin7 is reporting: <--MeterTeros11
    >>> 1D0! <--MeterTeros11
  Receiving results from MeterTeros11 at SDI12-1_Pin7 <--MeterTeros11
  Dialectric E: 1.24 <--MeterTeros11
  Temperature: 22.80 <--MeterTeros11
  Volumetric Water Content: 0.00 <--MeterTeros11
MeterTeros11 at SDI12-6_Pin7 is reporting: <--MeterTeros11
    >>> 6D0! <--MeterTeros11
  Receiving results from MeterTeros11 at SDI12-6_Pin7 <--MeterTeros11
  Dialectric E: 1.25 <--MeterTeros11
  Temperature: 22.80 <--MeterTeros11
  Volumetric Water Content: 0.00 <--MeterTeros11
2 results expected This differs from the sensor's standard design of 3 measurements!!
2 results expected This differs from the sensor's standard design of 3 measurements!!
MeterTeros11 at SDI12-1_Pin7 is reporting: <--MeterTeros11
    >>> 1D0! <--MeterTeros11
  Receiving results from MeterTeros11 at SDI12-1_Pin7 <--MeterTeros11
  Dialectric E: 1.23 <--MeterTeros11
  Temperature: 22.80 <--MeterTeros11
  Volumetric Water Content: 0.00 <--MeterTeros11
MeterTeros11 at SDI12-6_Pin7 is reporting: <--MeterTeros11
    >>> 6D0! <--MeterTeros11
[... 170 lines cut ...]
  Temperature: 22.80 <--MeterTeros11
  Volumetric Water Content: 0.00 <--MeterTeros11
MeterTeros11 at SDI12-6_Pin7 is reporting: <--MeterTeros11
    >>> 6D0! <--MeterTeros11
  Receiving results from MeterTeros11 at SDI12-6_Pin7 <--MeterTeros11
  Dialectric E: 11.11 <--MeterTeros11
  Temperature: 23.10 <--MeterTeros11
  Volumetric Water Content: 20.95 <--MeterTeros11
2 results expected This differs from the sensor's standard design of 3 measurements!!
2 results expected This differs from the sensor's standard design of 3 measurements!!
MeterTeros11 at SDI12-1_Pin7 is reporting: <--MeterTeros11
    >>> 1D0! <--MeterTeros11
  Receiving results from MeterTeros11 at SDI12-1_Pin7 <--MeterTeros11
  Dialectric E: 1.25 <--MeterTeros11
  Temperature: 22.80 <--MeterTeros11
  Volumetric Water Content: 0.00 <--MeterTeros11
MeterTeros11 at SDI12-6_Pin7 is reporting: <--MeterTeros11
    >>> 6D0! <--MeterTeros11
  Receiving results from MeterTeros11 at SDI12-6_Pin7 <--MeterTeros11
  Dialectric E: 11.08 <--MeterTeros11
  Temperature: 23.10 <--MeterTeros11
  Volumetric Water Content: 20.88 <--MeterTeros11

 \/---- Line Saved to SD Card ----\/
2000-06-06 04:30:00,1.25221,22.80,0.000,23.55,44.005,101441.46,4.806,26.00,11.13599,23.10,20.990

GArrigotti-cws, 2021-05-24:
@SRGDamia1 With the variable array debug:

> Executing task: C:\Users\ArrigottiG\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM4  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

Now running Arrigotti_NoRadio.ino on Logger Arrigotti

Using ModularSensors Library version 0.28.4
Logger timezone is set to UTC -8
RTC timezone is set to UTC      
Current RTC time is: 2000-06-06T04:33:08-08:00
All variable UUID's appear to be correctly formed.

2a5f077b-8de9-45ee-9704-4743726101fb -> SoilEa
168081de-2c1a-458f-8f86-1d88f7b8c394 -> SoilTemp
d7a3e8c8-d1f0-4503-a76f-93f5b6861850 -> SoilVWC
e5d304df-995b-4272-9a26-f3b525c315b7 -> BoschBME280Temp        
f8984e98-bbd4-4c29-a178-a64bcd546411 -> BoschBME280Humidity    
6872a270-5dfb-4cee-9326-118adc97af9e -> BoschBME280Pressure    
c3a10820-59ef-48a0-9fa5-88bb06914186 -> Battery
61e38606-b91c-4058-ac8c-3ff1ff280ede -> BoardTemp
29abed47-770f-44ef-893e-bff2645a1ff9 -> SoilEa
aa142833-d52e-4f91-b2b7-d3fa953acdae -> SoilTemp
49d31d59-1986-4a9c-ab6f-16d7639c8abd -> SoilVWC

There are 0 calculated variables in the group. <--VariableArray
There are 0 calculated variables in the group. <--VariableArray
This logger has a variable array with 11 variables, of which 11 come from 5 sensors and 0 are calculated.
Logger portion of setup finished.

Setting up sensors...
Beginning setup for sensors and variables... <--VariableArray
Running sensor setup functions. <--VariableArray
    EnviroDIY Mayfly at EnviroDIY Mayfly was already set up! <--VariableArray
    Set up of MeterTeros11 at SDI12-1_Pin7 ... <--VariableArray
        ... setup succeeded. <--VariableArray
    Set up of BoschBME280 at I2C_0x76 ... <--VariableArray
        ... setup succeeded. <--VariableArray
    Set up of MaximDS3231 at I2C_0x68 ... <--VariableArray
        ... setup succeeded. <--VariableArray
    Set up of MeterTeros11 at SDI12-6_Pin7 ... <--VariableArray
        ... setup succeeded. <--VariableArray
... Success! <--VariableArray
Setting up file on SD card
Data will be saved as Arrigotti_2000-06-06.csv
Putting processor to sleep

----- WARNING ----- !!!!!!!!!!!!!!!!!!!!
!!!!!!!!!! ----- WARNING ----- !!!!!!!!!!
!!!!!!!!!!!!!!!!!!!! ----- WARNING ----- 
 
The current clock timestamp is not valid!
 
----- WARNING ----- !!!!!!!!!!!!!!!!!!!!
!!!!!!!!!! ----- WARNING ----- !!!!!!!!!!
!!!!!!!!!!!!!!!!!!!! ----- WARNING ----- 
------------------------------------------
Creating a mask array with the uniqueness for each sensor.. <--VariableArray
Creating an array for the number of completed measurements.. <--VariableArray
Creating an array with the number of measurements to average.. <--VariableArray
Creating an array of the power pins.. <--VariableArray
Creating arrays of the power pin locations.. <--VariableArray
----->> Clearing all results arrays before taking new measurements. ... <--VariableArray
   ... Complete. <<----- <--VariableArray
----->> Powering up all sensors together. ... <--VariableArray
Powering up sensors... <--VariableArray
[... 195 lines cut ...]
10 . 3 --->> Collected result of reading 3 from MeterTeros11 at SDI12-6_Pin7 ... <--VariableArray
   ... got measurement result. <<--- 10 . 3 <--VariableArray
10 --->> Finished all measurements from MeterTeros11 at SDI12-6_Pin7 , putting it to sleep. ... <--VariableArray
   ... succeeded in putting sensor to sleep. <<--- 10 <--VariableArray
2 --->> MeterTeros11 at SDI12-1_Pin7 powered down. <<--- 2 <--VariableArray
5 --->> BoschBME280 at I2C_0x76 powered down. <<--- 5 <--VariableArray
10 --->> MeterTeros11 at SDI12-6_Pin7 powered down. <<--- 10 <--VariableArray
*****--- 5 sensors now complete ---***** <--VariableArray
----->> Averaging results and notifying all variables. ... <--VariableArray
--- Averaging results from MeterTeros11 at SDI12-1_Pin7 --- <--VariableArray
--- Notifying variables from MeterTeros11 at SDI12-1_Pin7 --- <--VariableArray
--- Averaging results from BoschBME280 at I2C_0x76 --- <--VariableArray
--- Notifying variables from BoschBME280 at I2C_0x76 --- <--VariableArray
--- Averaging results from EnviroDIY Mayfly at EnviroDIY Mayfly --- <--VariableArray
--- Notifying variables from EnviroDIY Mayfly at EnviroDIY Mayfly --- <--VariableArray
--- Averaging results from MaximDS3231 at I2C_0x68 --- <--VariableArray
--- Notifying variables from MaximDS3231 at I2C_0x68 --- <--VariableArray
--- Averaging results from MeterTeros11 at SDI12-6_Pin7 --- <--VariableArray
--- Notifying variables from MeterTeros11 at SDI12-6_Pin7 --- <--VariableArray
... Complete. <<----- <--VariableArray

 \/---- Line Saved to SD Card ----\/
2000-06-06 04:35:00,1.25552,22.90,0.000,23.56,44.423,101436.40,4.852,26.25,1.23408,23.70,0.000

------------------------------------------

GArrigotti-cws, 2021-05-24:
@SRGDamia1 Here are all three separate logs.

[Deep and Terrow.txt](https://github.com/EnviroDIY/ModularSensors/files/6534980/Deep.and.Terrow.txt)
[Variable Array Debug.txt](https://github.com/EnviroDIY/ModularSensors/files/6534981/Variable.Array.Debug.txt)
[Just Meter Teros.txt](https://github.com/EnviroDIY/ModularSensors/files/6534982/Just.Meter.Teros.txt)

SRGDamia1, 2021-05-24:
@GArrigotti-cws I'm sorry, I really want two logs, 

- one with **_3_** debug flags (`-DMS_SDI12SENSORS_DEBUG` **_and_** `-DMS_SDI12SENSORS_DEBUG` **_and_** -DMS_SDI12SENSORS_DEBUG_DEEP`) 

- and the second with **_4_** debug flags (`-DMS_SDI12SENSORS_DEBUG` **_and_** `-DMS_SDI12SENSORS_DEBUG` **_and_** `-DMS_SDI12SENSORS_DEBUG_DEEP` **_and_** `-DMS_VARIABLEARRAY_DEBUG`)

I want to see the raw SDI-12 commands and responses both with and without the variable array debug.  What you posted with the variable array debug doesn't have the SDI-12 command/response, so I can't see what's different.

SRGDamia1, 2021-05-24:
Looking at the "Just Meter Teros" file, it looks like everything worked fine.  Both sensors responded separately and you got different values in your csv for each.

GArrigotti-cws, 2021-05-24:
@SRGDamia1 The other file you requested.

[Variable Array Debug.txt](https://github.com/EnviroDIY/ModularSensors/files/6535037/Variable.Array.Debug.txt)

Yeah, I noticed that.  Not entirely sure why though, the same code is having very different outputs.

SRGDamia1, 2021-05-24:
The problem isn't with the sensors not responding, it's with the volumetric water content.

GArrigotti-cws, 2021-05-24:
@SRGDamia1 But if I disable the debugging libraries then output is very, very different for the results saved to the SD Card.  Same everything, 

2000-06-06 05:39:00,1.25123,23.10,0.000,23.63,44.978,101405.84,4.882,24.75,-9999.00000,-9999.00,0.000
2000-06-06 05:40:00,1.25247,23.10,0.000,23.45,45.646,101403.51,4.852,24.75,-9999.00000,-9999.00,0.000
2000-06-06 05:41:00,1.24670,23.10,0.000,23.40,45.535,101404.56,4.852,24.75,-9999.00000,-9999.00,0.000

aufdenkampe, 2021-05-24:
@GArrigotti-cws and @SRGDamia1, thanks for working on this!

Greg, Sara is asking for you to share with her the debug output when it fails. That's what she needs to see to figure out what is wrong.

Run it once with your `platformio.ini` file build flags like this, and save output to a text file called `4DebugFlags.txt`.

```ini
build_flags =
    -D SDI12_EXTERNAL_PCINT
    -D NEOSWSERIAL_EXTERNAL_PCINT
    -D TINY_GSM_RX_BUFFER=64
    -D TINY_GSM_YIELD_MS=2
    -D MQTT_MAX_PACKET_SIZE=240
    -DMS_METERTEROS11_DEBUG
    -DMS_SDI12SENSORS_DEBUG 
    -DMS_SDI12SENSORS_DEBUG_DEEP
    -DMS_VARIABLEARRAY_DEBUG
```

Then run it again with the last line commented out, and save output to a text file called `3DebugFlags.txt`.

```ini
build_flags =
    -D SDI12_EXTERNAL_PCINT
    -D NEOSWSERIAL_EXTERNAL_PCINT
    -D TINY_GSM_RX_BUFFER=64
    -D TINY_GSM_YIELD_MS=2
    -D MQTT_MAX_PACKET_SIZE=240
    -DMS_METERTEROS11_DEBUG
    -DMS_SDI12SENSORS_DEBUG 
    -DMS_SDI12SENSORS_DEBUG_DEEP
    ; -DMS_VARIABLEARRAY_DEBUG
```

SRGDamia1, 2021-05-24:
Ok, I have some suspicions about what might be causing the failures.  I think it's probably something in the VWC calculation that's throwing everything else off.   To fix it, I'm going to modify the Teros to output *4* variables - keeping the "calibratedCountsVWC" that is currently being ditched and I'm going to separate out the calculations a little.  I'll try to get something ready for you to test tonight.

aufdenkampe, 2021-05-24:
@SRGDamia1, thank you!

I'm on the phone with Greg right now, and he says that enabling `-DMS_SDI12SENSORS_DEBUG_DEEP` also causes both sensors to work. We had previously discovered that enabling `-DMS_VARIABLEARRAY_DEBUG` also causes both sensors to work.

So it makes it hard to see what is going on to cause the failures.

GArrigotti-cws, 2021-05-24:
@SRGDamia1 @aufdenkampe   I can reproduce a failure by doing the following:

```ini
[build]
build_flags =
    -DSDI12_EXTERNAL_PCINT
    -DNEOSWSERIAL_EXTERNAL_PCINT
    -DMQTT_MAX_PACKET_SIZE=240
    -DTINY_GSM_RX_BUFFER=64
    -DTINY_GSM_YIELD_MS=2
    -D MS_SDI12SENSORS_DEBUG    
```

[Failure.txt](https://github.com/EnviroDIY/ModularSensors/files/6535271/Failure.txt)

aufdenkampe, 2021-05-24:
@SRGDamia1, Greg says the test sensors are in the air, so volumetric water content should be zero. That might not be the problem.

aufdenkampe, 2021-05-24:
@SRGDamia1, I just noticed that you released Modular Sensors v0.28.5 two weeks ago.

Greg and I have been using v0.28.4. Any chance that might be a factor?

GArrigotti-cws, 2021-05-25:
@aufdenkampe Looks like the latest modular sensors still did not rectify the problem, ironically it inverted which sensor had -9999.

'''terminal
2000-06-07 00:27:00,-9999.00000,-9999.00,0.000,23.13,44.730,101464.29,4.867,24.75,1.21920,23.10,0.000
```

My .ino file:

```ini
[external:libraries]
lib_deps =
    EnviroDIY_ModularSensors@=0.28.5
    https://github.com/PaulStoffregen/AltSoftSerial.git
    https://github.com/SRGDamia1/NeoSWSerial.git
    https://github.com/EnviroDIY/SoftwareSerial_ExternalInts.git
    https://github.com/vshymanskyy/StreamDebugger.git

[build]
build_flags =
    -DSDI12_EXTERNAL_PCINT
    -DNEOSWSERIAL_EXTERNAL_PCINT
    -DMQTT_MAX_PACKET_SIZE=240
    -DTINY_GSM_RX_BUFFER=64
    -DTINY_GSM_YIELD_MS=2
    -D MS_SDI12SENSORS_DEBUG    
```

The .cpp file is as follows:

```cpp

const char*   teros11SDI12address = "1"; 
const int8_t  terosPower = sensorPowerPin;  
const int8_t  terosData  = 7;            
const uint8_t teros11NumberReadings = 3;   

MeterTeros11 teros11(*teros11SDI12address, terosPower, terosData, teros11NumberReadings);

const char*   teros11_2SDI12address = "6"; 
MeterTeros11 teros11_2(*teros11_2SDI12address, terosPower, terosData, teros11NumberReadings);
```

These are the timings I tried with the new modular sensors:

```cpp
#define TEROS11_WARM_UP_TIME_MS 250
#define TEROS11_STABILIZATION_TIME_MS 50
#define TEROS11_MEASUREMENT_TIME_MS 50
#define TEROS11_EXTRA_WAKE_TIME_MS 0
```

```cpp
#define TEROS11_WARM_UP_TIME_MS 1550
#define TEROS11_STABILIZATION_TIME_MS 50
#define TEROS11_MEASUREMENT_TIME_MS 50
#define TEROS11_EXTRA_WAKE_TIME_MS 0
```

I verified against the new modular sensors, same results.

SRGDamia1, 2021-05-25:
Can you try the new sensor_calc branch:  https://github.com/EnviroDIY/ModularSensors/tree/sensor_calc?

I added an internal parameter for sensors referring to the number of variables that are calculated internally and then split out the calculations for the Teros some.  I also added some debugging.

Can you post results with no debugging and with all the SDI-12 debugging?
-

SRGDamia1, 2021-05-25:
you should be able to pull the new branch with:
```ini
lib_deps =
    https://github.com/EnviroDIY/ModularSensors.git#sensor_calc
```

GArrigotti-cws, 2021-05-25:
@SRGDamia1 Would you like the deep debugging enabled?  Or the normal SDI12?

GArrigotti-cws, 2021-05-25:
@SRGDamia1 After some troubleshooting, the following output log was produced:

[Sensor_Calc.txt](https://github.com/EnviroDIY/ModularSensors/files/6542383/Sensor_Calc.txt)

```ini
[external:libraries]
lib_deps =
    https://github.com/EnviroDIY/ModularSensors.git#sensor_calc
    https://github.com/PaulStoffregen/AltSoftSerial.git
    https://github.com/SRGDamia1/NeoSWSerial.git
    https://github.com/EnviroDIY/SoftwareSerial_ExternalInts.git
    https://github.com/vshymanskyy/StreamDebugger.git

[build]
build_flags =
    -DSDI12_EXTERNAL_PCINT
    -DNEOSWSERIAL_EXTERNAL_PCINT
    -DMQTT_MAX_PACKET_SIZE=240
    -DTINY_GSM_RX_BUFFER=64
    -DTINY_GSM_YIELD_MS=2
    -D MS_SDI12SENSORS_DEBUG    
    -DMS_SDI12SENSORS_DEBUG_DEEP 
```

```terminal
2000-06-07 05:25:00,1.26650,22.20,0.000,22.60,41.964,101565.40,4.882,24.00,1.22685,22.40,0.000
```

Based on several outputs they both appear to be working with the above configuration.  However, if I turn off the debugging:

```ini
    -D MS_SDI12SENSORS_DEBUG    
    -DMS_SDI12SENSORS_DEBUG_DEEP 
```

Then I receive this output:

```terminal
2000-06-07 05:32:00,-9999.00000,-9999.00,-9999.000,22.69,41.926,101554.37,4.882,24.00,1.22309,22.50,0.000
```

SRGDamia1, 2021-05-25:
Okay.  I'm flat-out stumped.  I made some tiny changes, which you should be able to get with a `pio lib update`, but they shouldn't make a difference.

Get another Mayfly and program it with this freshly-written SDI-12 spy program:  https://github.com/EnviroDIY/Arduino-SDI-12/blob/master/tools/SDI12_spy/SDI12_spy.ino

Connect the ground and sdi-12 data (7) pins, keeping your wires as short as possible.  Post the output from the spy along with the output from running with and without the debugging flags.  There probably will be some garbage in the output from the spy, especially if your grounds aren't perfect, but it should at least give us some idea of what's going on.

SRGDamia1, 2021-05-25:
There's definitely going to be garbage in the spy line, because it will parse the controlling line changes as garbage characters.  I'll think about fixing that.  But, anyway, even with garbage, it might help.

GArrigotti-cws, 2021-05-25:
@SRGDamia1 I've got to buy some additional cables, but I'll do that tonight and attempt in the morning.

SRGDamia1, 2021-05-25:
https://github.com/EnviroDIY/Arduino-SDI-12/issues/77

GArrigotti-cws, 2021-07-06:
@SRGDamia1 I'm hoping to get those cables to provide the information, I apologize for the delay.  Had to shift priorities temporarily.

aufdenkampe, 2021-07-08:
@SRGDamia1, we're back at this issue, and now I have two Teros11 sensors for testing.

For some reason, I'm seeing the error even when the relevant debug statements are turned on!

What I see is that within an action cycle, the first of the two SDI-12 sensors does not respond  on the first try, but does on the second. We see this during setup with `requestSensorAcknowledgement`, but we try a second time and it works. However, we don't try to "Collect result of reading" a second time (i.e. the SDI12 command is sent only once per cycle). Therefore we fail.

This is the specific output sections I'm referring to:
```
9 . 1 --->> Starting reading 1 on MeterTeros11 at SDI12-6_Pin7 ... <--VariableArray
  Asking for sensor acknowlegement <--SDI12Sensors
    >>> 6! <--SDI12Sensors
    <<< ␇6 <--SDI12Sensors
    MeterTeros11 at SDI12-6_Pin7 did not reply! <--SDI12Sensors
    >>> 6! <--SDI12Sensors
    <<< 6 <--SDI12Sensors
    MeterTeros11 at SDI12-6_Pin7 replied as expected. <--SDI12Sensors
```

Then later...

```
9 . 1 --->> Collected result of reading 1 from MeterTeros11 at SDI12-6_Pin7 ... <--VariableArray
MeterTeros11 at SDI12-6_Pin7 is reporting: <--MeterTeros11
    >>> 6D0! <--SDI12Sensors
    <<< 6 <--SDI12Sensors
 <--SDI12Sensors
    <<< -9999.0000000000 <--MeterTeros11
    <<< 
```

DETAILS:

The output below is compiled with:
```ini
build_flags =
    -D MS_SDI12SENSORS_DEBUG
    -D MS_SDI12SENSORS_DEBUG_DEEP
    -D MS_METERTEROS11_DEBUG
    -D MS_VARIABLEARRAY_DEBUG

lib_deps =
    https://github.com/EnviroDIY/ModularSensors.git#sensor_calc
```

```
Now running CWS-ThomasDairy1.ino on Logger CWS-ThomasDairy1

Using ModularSensors Library version 0.30.0
TinyGSM Library version 0.11.4

Logger timezone is set to UTC -8
RTC timezone is set to UTC
Current RTC time is: 2021-07-08T08:26:55-08:00
All variable UUID's appear to be correctly formed.

6b3a0e43-3392-4e83-b0aa-ed236b5d9ae9 -> Battery
db7d4a18-5b98-4188-bf7d-8762cb1da2d3 -> SoilEa
9da8d39f-710d-4678-b1f3-009f667a7073 -> SoilTemp
8c26d5d9-da81-4218-a198-a78c815cedce -> SoilVWC
fe553ef8-75cd-4677-9f01-82d9bbce3762 -> BoschBME280Temp
63401eba-984b-4b34-856d-9c0cbc9d9a48 -> BoschBME280Humidity
99b87310-03aa-40ca-beab-2696e3db22bb -> BoschBME280Pressure
858b111c-014b-40ad-bf24-4aa15c35f1ef -> SoilEa
929ca303-01b3-49a8-a96b-17121c8cb8c7 -> SoilTemp
f68e2659-f2bf-492e-a4d6-2313e825dc61 -> SoilVWC
c058446f-1f3f-4ece-b8a3-7cb88ef9953b -> signalPercent
 
There are 1 calculated variables in the group. <--VariableArray
There are 1 calculated variables in the group. <--VariableArray
This logger has a variable array with 11 variables, of which 10 come from 4 sensors and 1 are calculated.
[... 252 lines cut ...]
6 . 3 --->> Starting reading 3 on BoschBME280 at I2C_0x76 ... <--VariableArray
   ... start reading succeeded. <<--- 6 . 3 <--VariableArray
6 . 3 --->> Collected result of reading 3 from BoschBME280 at I2C_0x76 ... <--VariableArray
   ... got measurement result. <<--- 6 . 3 <--VariableArray
6 --->> Finished all measurements from BoschBME280 at I2C_0x76 , putting it to sleep. ... <--VariableArray
   ... succeeded in putting sensor to sleep. <<--- 6 <--VariableArray
3 --->> MeterTeros11 at SDI12-1_Pin7 powered down. <<--- 3 <--VariableArray
6 --->> BoschBME280 at I2C_0x76 powered down. <<--- 6 <--VariableArray
9 --->> MeterTeros11 at SDI12-6_Pin7 powered down. <<--- 9 <--VariableArray
*****--- 4 sensors now complete ---***** <--VariableArray
----->> Averaging results and notifying all variables. ... <--VariableArray
--- Averaging results from EnviroDIY Mayfly at EnviroDIY Mayfly --- <--VariableArray
--- Notifying variables from EnviroDIY Mayfly at EnviroDIY Mayfly --- <--VariableArray
--- Averaging results from MeterTeros11 at SDI12-1_Pin7 --- <--VariableArray
--- Notifying variables from MeterTeros11 at SDI12-1_Pin7 --- <--VariableArray
--- Averaging results from BoschBME280 at I2C_0x76 --- <--VariableArray
--- Notifying variables from BoschBME280 at I2C_0x76 --- <--VariableArray
--- Averaging results from MeterTeros11 at SDI12-6_Pin7 --- <--VariableArray
--- Notifying variables from MeterTeros11 at SDI12-6_Pin7 --- <--VariableArray
... Complete. <<----- <--VariableArray

 \/---- Line Saved to SD Card ----\/
2021-07-08 08:30:00,4.867,1.20816,26.40,0.000,25.00,44.382,97955.52,-9999.00000,-9999.00,-9999.000,-9999

```

aufdenkampe, 2021-07-08:
I found the fix. 

```ini
build_flags =
    -D MS_SDI12_NON_CONCURRENT
```

Evidently the Teros 11 sensors do not like concurrent SDI-12 measurements. Both of my sensors are supposedly `SDI12 Version: 1.30` but they do have different Meter firmware versions (112 & 114).

aufdenkampe, 2021-07-08:
... but defining `MS_SDI12_NON_CONCURRENT` does't work for the new ModularSensors v0.30.0! 
In fact, neither sensor reports.

aufdenkampe, 2021-11-19:
@SRGDamia1, thanks for merging `master` into the `sensor_calc` branch.
I'm interested in revisiting and finally fixing this issue over the next month or two. 

From my comment above, I may need your help in implementing the fix. I'll let you know when I'm ready for testing again.

SRGDamia1, 2021-11-20:
@aufdenkampe - do let me know, but I'm really, really behind on pretty much everything right now.  Before I left in July, I was starting to pick through the SDI-12 library to see if I could optionally use a different (10 or 16 bit) timer where possible instead of the 8bit timer to be able to have a little more buffer in reading the SDI-12 bits in before the timer rolls over.  I was thinking that might improve the back and forth just a touch.  But I haven't picked it up since then and I don't know when I'll get a chance to think about it again.

I've also dropped my hours just a little to better manage 4 kids at home.

aufdenkampe, 2021-11-24:
@SRGDamia1, thank you so much for issuing release v0.32.2!
https://github.com/EnviroDIY/ModularSensors/releases/tag/v0.32.2

It looks like you addressed many if not all the issues I raised here! Thank you!

I understand that your time is limited. I will let you know when I am able to test it all out.

aufdenkampe, 2023-09-19:
The Teros 11 sensors are working well as far as I know. I should have closed this a while ago!

closed 2023-09-19 by aufdenkampe

aufdenkampe, 2023-09-19:
It's possible that there is still a bug in this implementation. See:
- https://github.com/EnviroDIY/Arduino-SDI-12/issues/75

commit e08e388e (2019-08-28): Add METER TEROS 11 to menu example

This should be the final commit to close #276