EnviroDIY/ModularSensors #277: Switch examples to use Digi LTE-M Transparent mode

aufdenkampe, 2019-08-15:
It's probably time to switch our examples to use Digi XBee3 LTE-M radio module in transparent mode. Presently, only the [`DRWI_LTE.ino`](https://github.com/EnviroDIY/ModularSensors/blob/master/examples/DRWI_LTE/DRWI_LTE.ino) example uses transparent mode, and that approach seems to be working well for a number of us.

### Background:
Our new [LTEbee-Adapter](https://github.com/EnviroDIY/LTEbee-Adapter) allows us to use Transparent mode for the Digi XBee3 LTE-M module, without brown-outs.  Previously, we could only get Bypass mode to work well, likely because of less stringent power demands. In fact, with the current version of the library (v0.23.4) it now appears that Transparent mode is more stable than Bypass mode, and recommended, even thought it still has some draw backs.

@SRGDamia1 summarized her latest findings here: https://github.com/EnviroDIY/ModularSensors/commit/1f8615dffefea1ffe592ac4f90092fd78880b724#commitcomment-34592670

> At the beginning of the summer I finally got some boards and SIM cards to test the LTE a bit more "long term" on my desk.  Running in bypass mode seemed to send the data more consistently and to use up less battery.  I was all ready to confirm that bypass was better, when the board completely locked up - all the lights on and doing nothing.  So I reprogrammed it to turn on more of the debugging and let it go again.. and it locked up after a few days.  I could never get it to lock up "on queue" and still can't figure out why it's locking up.  Every time, hitting the reset button magically fixes all ills.  The transparent board never locked up, just ran out of battery faster and had less success in sending data.  I decided that was a better trade off.  In the process I also figured out and implemented a watch dog timer to force a reset in event of a freeze up while the power is on.
> 
> Summary:
> Transparent
> \+ hasn't been observed to cause mcu to fully lock up
> \+ uses very slightly less data with every data transmission
> \- doesn't make a connection as quickly
> \- uses more battery power because of slower connection
> \- completely misses more data points (you can see many of them on hologram as 0-data connections)
> 
> Bypass
> \+ connects faster = lower power
> \+ almost every point makes it to the web
> \- uses slightly more data with each individual transmission (this is in overhead we cannot control)
> \- sometimes causes everything to freeze up while fully powered - relies on the watch-dog timer to prevent full shutdown.

Considering my positive experiences deploying with Transparent mode in recent weeks, and  comments from @neilh10 in https://github.com/EnviroDIY/ModularSensors/commit/1f8615dffefea1ffe592ac4f90092fd78880b724#commitcomment-34697982, let's make the switch in our examples and docs.

neilh10, 2019-08-15:
Thanks for the references to the LTEbee-adapter, I hadn't seen the git,  I've got an adapter on the way.  I've started using the DigiXBeeCellularTransparent/Mayfly with Digi LTE CatM1 XB3-C-A2  - with a hologram SIM - though working on trying to get it provision with the SIM.

One item I did for a piece of production equipment testing - was to read and print the SIM #. This verifies the basics all work - from Digi module seated to small SIMs readable.

SRGDamia1, 2019-08-28:
@aufdenkampe - FYI, while I haven't been able to figure out why the bypass mode sometimes completely freezes up the board, the watch-dog timer *does successfully restart the board* when it is frozen.  It's probably a bit more risky depending on the watch-dog timer like that, but on the other hand, that's exactly what the watch-dog is designed for.  

So, you're trading dependence on the watch-dog timer and higher data use for more successful transmissions and less battery draw - not freezes completely vs doesn't freeze up.  For the ones we've got going out this week, @s-hicks2 thinks it's probably worth it.

I haven't run a long enough test to see the watch-dog come into effect for other reasons, but I'm hoping that it will reduce the number of times that stations need to be visited and power cycling them is a magic fix.

commit 3693b769 (2019-09-19): example to transparent, closes #277
closed 2019-09-19 by SRGDamia1