EnviroDIY/ModularSensors #278: DigiXBeeCellularTransparent setup of XBC LTE-M/NB-IoT Global

neilh10, 2019-08-15:
I'm working through getting a Hologram sim connected to Digi LTE-M XB3-C-A2 /Ltee Module/Mayfly 0.5b. 
I'm using the envirodiy: master  XBeeCellularTransparent and monitoring AT cmds being issued.
Its in California with both ATT and Verizon.
I've upgraded the XB3-C-A2 to the latest firmware  thru X-CTU, 
Which says
Updated firmware to 1140A
XBC LTE-M/NB-IoT Global
IMEI: 352753090120971

It doesn't appear to be finding a network so just checking on the sequence of cmds I'm seeing.

Specifically when looking for a cellular network, its issuing ATAI  and getting from the XB3 a response 22. Its not happy with 22 and continues issuing them at 0.25seconds.
I can't find this command "AI" in the Users Guide manual  
Digi Xbee3 Cell LTE-M 90002258 revk 2019July .pdf

I wonder if there are any insights that can be shared?. Have I got the right approach?. 
There could be different conditions here with the network so I guess I need to show some network information.

------log file.
Power to unspecified modem is not controlled by this library. <--LoggerModem
  Modem setup up 1st pass <--tu_mon
Running wake function for unspecified modem <--LoggerModem
Setting pin 23 LOW to wake XBee <--DigiXBee
unspecified modem should be awake. <--LoggerModem
  Connecting to the Internet...  <--tu_mon
Setting up the modem ... <--LoggerModem
Modem was already awake and should be ready for setup. <--LoggerModem
Running modem's begin function ... <--LoggerModem
Initializing the XBee... <--DigiXBeeCellularTransparent
+++OK
ATAP0
OK
ATGT64
OK
ATCT64
OK
ATWR
OK
ATAC
OK
ATHS
B08
ATCN
OK
Putting XBee into command mode... <--DigiXBeeCellularTransparent
+++OK
Setting I/O Pins... <--DigiXBeeCellularTransparent
ATD81
OK
ATD91
OK
ATD71
OK
ATD51
OK
ATP01
OK
Setting Sleep Options... <--DigiXBeeCellularTransparent
ATSM1
OK
ATSO0
OK
Setting Other Options... <--DigiXBeeCellularTransparent
ATDO0
OK
ATTD0A
OK
ATP10
OK
Setting Cellular Carrier Options... <--DigiXBeeCellularTransparent
ATCP2
OK
ATN#2
ERROR
ATANhologram
OK
ATWR
OK
ATAC
OK
Ensuring XBee is in transparent mode... <--DigiXBeeCellularTransparent
ATAP1
OK
Applying changes... <--DigiXBeeCellularTransparent
ATWR
OK
ATAC
OK
ATCN
OK
Restarting XBee... <--DigiXBeeCellularTransparent
+++OK
ATAM1
OK
ATWR
OK
ATAC
OK
ATFR
OK
++++++~
u+++OK
ATAM0
OK
ATWR
OK
ATAC
OK
ATCN
OK
+++OK
ATAP0
OK
ATGT64
OK
ATCT64
OK
ATWR
OK
ATAC
OK
ATHS
B08
ATCN
OK
... Setup successful! <--DigiXBeeCellularTransparent
... Complete!  It's a Digi XBee3 Cellular LTE-M <--LoggerModem
Digi XBee3 Cellular LTE-M warms up in 0 ms, indicates status in 15 ms, is responsive to AT commands in less than 5000 ms, and takes up to 15000 ms to close connections and shut down. <--LoggerModem
Leaving modem on after setup ... <--LoggerModem

Waiting for Digi XBee3 Cellular LTE-M to respond to AT commands... <--DigiXBeeCellularTransparent
+++OK
ATCN
OK
... AT OK after 496 milliseconds! <--DigiXBeeCellularTransparent

Waiting up to 50 seconds for internet availability... <--DigiXBeeCellularTransparent
+++OK
ATAI
22
ATAI
22
ATAI
22
ATAI
22
ATAI
22
ATAI

SRGDamia1, 2019-08-16:
What underlying ublox firmware do you have?  I've had it work with every version of the firmware, but the current (11413 with L0.0.00.00.05.08 App version 02.04) is the best. 

Can you successfully connect to the network when using XCTU? You may need to wait a long time to get connection the first time after configuration/firmware/location changes - up to 20 minutes, or even more.  ModularSensors doesn't keep the modem on that long in the beginning. 

I've never seen hologram connect to Verizon, even in places where the Verizon signal is stronger than AT&T. Again, test with xctu and automatic or Verizon band selection to see what you get.

SRGDamia1, 2019-08-16:
Digi doesn't provide any commands to list the operators/networks available. You can use bypass mode or USB correct and run AT+COPS?.

neilh10, 2019-08-16:
Hey thanks for the tips ( I'm getting some time on this). 

Interesting  u-blox SARA-R410M is  separate from the firmware release.  
I have upgraded to Version 11413 August 7, 2019  and indicates XCTU MV modem firmware Version L0.0.00.00.05.06,A.02.00
(The release notes 11413 says  new u-blox release downloaded separately, It INCLUDES support for u-blox SARA-R410M-02B-01 version [L0.0.00.00.05.08 ](https://www.u-blox.com/sites/default/files/SARA-R410M-02B-01_PCN_%28UBX-19024506%29.pdf )but needs to be programmed separately. Release note don't seem to have anything critical for operation, but says needed for regulatory approval process, so implies that is likely to be final release in couple of months )

I left  XB3 running with XCTU and AN set to "hologram", and after 10minutes refreshed the "MN Operator" and it came back with "310 410 Hologram"
Also then refreshed the MY Module UP Address and came back with "10.179.34.250"
- thanks for the tip.
So timing just on XB3/XCTU
2nd boot then said "AT&T Hologram" almost straight away (didn't time ~1min)
3rd boot - timed it to 1min - "310 410 Hologram"
4th boot 1min 20 to "310 410 Hologram"
5th boot 59sec to "AT&T Hologram" - this time I alternated between requesting DB and MN.  DB takes time, and then when has a signal can connect on MN. AI was too far down screens to be able to poll from XCTU - I think it should be scan for DB reading,( not empty string), and then look for AI and then can expect MN. 
AI is the correct indication but doesn't help with debugging location setup if not seeing a signal, DB gives a reading whether or not there is a local network found.
With the older G3 systems (on multi-tech modems) I saw it take 1minute to find a cellphone tower, before data was able to pass.
I believe with the new CAT-M1  once the network tower has been  synchronized, even powering down results in keeping a sleeping association with the cell tower for some time. Not sure which release.
However this a significant power saving as passing data is pretty quick compared to 1minute startup

 
Release 11411(April 12, 2019) says 
3. Active Scan command (ATAS)
o Scans for mobile cells in the vicinity and returns information about the cells in the service area of
the XBee.
However X-CTU latest (6.4.3) doesn't appear to support AS it.

SRGDamia1, 2019-08-16:
I'd missed the ```ATAS``` command. Thanks.  I need to add it to TinyGSM.  https://github.com/vshymanskyy/TinyGSM/issues/321

Operator "310 410" is AT&T.

SRGDamia1, 2019-08-16:
You absolutely should upgrade to the u-blox firmware to at least A.02.01.  It was a kind-of painful update that involved using MicroPython on the module.  Digi really stresses the importance of that firmware update.

The update to A.02.04 was much less painful with USB direct via u-blox's own EasyFlash program.  The firmware is here:  ftp://ftp1.digi.com/support/firmware/u-blox_EasyFlash_10.02_SARA-R410M-02B-01_IP.zip.  I don't know if you can go straight to A.02.04 without going through A.02.01.  I'd already done all the updates before they started supporting the newer version.  I don't think Digi is supporting the A.02.04 update without USB direct because they don't think it's as crucial.  So to get it you'll need a way to connect the USB direct pins to your computer.

neilh10, 2019-08-17:
Ok - hadn't twigged there was another level of device software. 
Turns out there is an API Over the Wire method of updating to get  L0.0.00.00.05.06,A.02.01

90002258revk Manual Pg 35
https://www.digi.com/support/productdetail?pid=5635&type=firmware
Per "U-blox Over the Wire Update Script"
modules\Digi-MaxStream\Xbee3\over-the-wire

(windows10 prof:  pip version upgrade to  19.2.2 )
$ pip install virtualenv
$ virtualenv env
$ env\Scripts\activate.bat
(env) .. modules\Digi-MaxStream\Xbee3\over-the-wire> pip install -r requirements.txt
(With XCTU board on COM10)
$ python update.py COM10
(then takes 20+ minutes)

$ python update.py COM10
2019-08-16 16:49:01,285 update.py: INFO: Starting communication with device: COM10, baud rate 9600
2019-08-16 16:49:01,473 update.py: INFO: Communication with the XBee3 Cellular device is working!
2019-08-16 16:49:01,629 update.py: INFO: IMEI of XBee3 Cellular device: 352753090120971
2019-08-16 16:49:01,629 update.py: INFO: This is an 'ES2' device.
2019-08-16 16:49:01,629 update.py: INFO: Reading cellular component version string (ATMV)...
2019-08-16 16:49:01,676 update.py: INFO: MV: L0.0.00.00.05.06,A.02.00
2019-08-16 16:49:01,741 update.py: INFO: Current BD: 3. Current RTS/D6: 0. Current CTS/D7: 1.
2019-08-16 16:49:01,850 update.py: INFO: Attempting stage 1 update using 2stg-upd-for-ES2/stg1-pkg1-5.6A2.00_to_5.6A2.01.pkg
2019-08-16 16:49:01,850 urllib3.connectionpool: DEBUG: Starting new HTTP connection (1): ftp1.digi.com:80
2019-08-16 16:49:02,298 urllib3.connectionpool: DEBUG: http://ftp1.digi.com:80 "GET /support/ublox/2stg-upd-for-ES2/stg1-pkg1-5.6A2.00_to_5.6A2.01.pkg HTTP/1.1" 200 2686
2019-08-16 16:49:02,298 update.py: INFO: Rebooting the XBee3 Cellular device to initiate update.
2019-08-16 16:49:02,298 update.py: INFO: Performing clean shutdown of cellular component...
2019-08-16 16:49:38,508 update.py: INFO: Resetting the XBee3 Cellular device...
2019-08-16 16:49:38,806 update.py: INFO: Waiting 60 seconds to allow the cellular component to initialize...
Delayed 60 seconds |#########################################################################################################################################################|
2019-08-16 16:50:39,239 update.py: INFO: Informing the XBee3 Cellular device that we are beginning an update.
2019-08-16 16:50:39,239 update.py: INFO: Update is starting.
2019-08-16 16:50:39,239 update.py: INFO: Note: The first chunk can take 30 seconds or more to complete.
Sending update file... 100%|###################################################################################################################################|Time:  0:00:20
2019-08-16 16:50:59,849 update.py: INFO: ... Sending final empty chunk, #3, to signal the end of the file and apply the upgrade. This can take as long as 20 minutes to complete.
2019-08-16 16:51:35,130 update.py: INFO: Successful stage 1 update using 2stg-upd-for-ES2/stg1-pkg1-5.6A2.00_to_5.6A2.01.pkg
2019-08-16 16:51:35,130 update.py: INFO: Attempting stage 2 update
2019-08-16 16:51:35,131 urllib3.connectionpool: DEBUG: Starting new HTTP connection (1): ftp1.digi.com:80
2019-08-16 16:51:38,785 urllib3.connectionpool: DEBUG: http://ftp1.digi.com:80 "GET /support/ublox/2stg-upd-for-ES2/stg2-pkg-5.6A2.00_to_5.6A2.01.pkg HTTP/1.1" 200 1580036
2019-08-16 16:51:38,794 update.py: INFO: Rebooting the XBee3 Cellular device to initiate update.
2019-08-16 16:51:38,795 update.py: INFO: Performing clean shutdown of cellular component...
2019-08-16 16:51:42,809 update.py: INFO: Resetting the XBee3 Cellular device...
2019-08-16 16:51:43,111 update.py: INFO: Waiting 60 seconds to allow the cellular component to initialize...
Delayed 60 seconds |#########################################################################################################################################################|
2019-08-16 16:52:43,520 update.py: INFO: Informing the XBee3 Cellular device that we are beginning an update.
2019-08-16 16:52:43,520 update.py: INFO: Update is starting.
2019-08-16 16:52:43,520 update.py: INFO: Note: The first chunk can take 30 seconds or more to complete.
Sending update file... 100%|###################################################################################################################################|Time:  0:04:52
2019-08-16 16:57:36,457 update.py: INFO: ... Sending final empty chunk, #1055, to signal the end of the file and apply the upgrade. This can take as long as 20 minutes to comp
lete.
2019-08-16 17:08:27,038 update.py: INFO: Reading cellular component version string (ATMV)...
2019-08-16 17:08:27,038 update.py: INFO: Update successful. New MV value: L0.0.00.00.05.06,A.02.01
2019-08-16 17:08:27,038 update.py: INFO: Cellular component update successful.
2019-08-16 17:08:27,053 update.py: INFO: Restoring previous baud rate and RTS/CTS configuration.
2019-08-16 17:08:27,277 update.py: INFO: Completed update of XBee3 Cellular device with IMEI 352753090120971
2019-08-16 17:08:27,277 update.py: INFO: Updates complete. This script should automatically exit in a moment. If it does not exit, use Ctrl-C to stop the script.

neilh10, 2019-08-17:
Success hurrah - one post including NIST attempt. Checking Hologram dashboard says 24.46 KB  - got through.

Though data.envirodiy  isn't happy yet....
-- Response Code --
504

piffle puffle!!!

neilh10, 2019-08-19:
So just an update - the "504" is internally defined timeout. Basically with the Post there is no response - effectively the TCP connection hasn't worked. 
I'm going back to basics of the modules to investigate.
The original question I asked was what version - and I think that baseline is  - 
11413 with L0.0.00.00.05.08 App version 02.04)
I'm at  11414 L0.0.00.00.05.06,A.02.01  and need to figure out how to upgrade.

Per the manual,  I'm trying the Echo Server - and that doesn't appear to be working.
I am getting the physical wireless registration - and Hologram/ATT are reporting data is being passed.
So it could be that my cellular signal strength is too weak (0x5D db) and I might need a better antenna (currently Laird UMTS Band1 from Digikey).

So I have a few action items to work through.

An observation (what I'm chewing on) - using the reference OSI 7 layer model - the upper layers managing the XB3 as a Network Layer Element - don't have any realtime visibility to the status of the physical wireless layer and also the TCP connection.  
If the physical wireless layer goes to something like AI=0x23 that doesn't become visible. 
The same is true of the TCP link, I haven't figured out a  way for checking status for the TCP being connected, and also if that link got reset should that be visible. 
When wireless works its fantastic - the general wireless problem is; when its on the edge of its range which for rural riparian geographical coverage  can be large, the system is likely to see the wire broken a lot - and the systems strength is in dealing with a link going  down and then coming back depending on external conditions - like the strength of the wind.   
I do remind myself in thinking about installers working in riparian situations feedback about whether it works at installation or not is pretty valuable - having done some installations on hot sweaty, poison ivy covered riverbanks .!! so some challenges. :) and as they say Rome wasn't built in a day.

neilh10, 2019-08-19:
Yeah - I got a good POST. I releaized as I was trying to switch to another task - I was staring at an Antenna connect for an outdoor antenna I had  put up for something else - roughly in the Cellular range - so what the heck should try it. 
Connected it to the XCB3 u.fl on the XCTU board. Tried the XCB3 Time of Day XCTU example - it worked, then the ECHO server XCTU example it worked. 
What the heck. 
Put the XB3 in the Mayfly . Powered. Response Code 201!! It worked .  Hurrah!!

neilh10, 2019-08-21:
Worked overnight no problem with a good signal strength.

closed 2019-08-21 by neilh10