EnviroDIY/ModularSensors #280: Auto lib crawl of library.json and rebuild  .pio\libdeps\mayfly failing

neilh10, 2019-08-20:
I'm just wondering has anybody had an incompatibility with platformio rel 4.0 and ModualrSensors\library.json

I just deleted my .pio\libdeps\mayfly and then attempted a rebuild It didn't pull in the manifest. 
This suggest it should
https://docs.platformio.org/en/latest/librarymanager/config.html#library-config
Though not sure I understand the subtley's of the libs as "dependencies"
https://docs.platformio.org/en/latest/librarymanager/config.html#dependencies

if I used "platformio lib install " it worked to pull them in.

Maybe I've set a switch wring

SRGDamia1, 2019-08-20:
It's an issue with the new release of PlatformIO. For some reason dependencies aren't being pulled. I think they already have an issue open over there.

neilh10, 2019-08-21:
Ok thanks. I

closed 2019-08-21 by neilh10