EnviroDIY/ModularSensors #281: support NTP where applicable

neilh10, 2019-08-21:
Synced time is an essential requirement for meaningful logging of environmental data on ModularSensors.
Accurate time is imparted to a system running ModularSensor through accessing NIST servers.
The https://tf.nist.gov/tf-cgi/servers.cgi encourages moving to network time protocol (NTP), which uses UDP.
While the current code uses an earlier NIST resource that runs over tcp/ip - this is likely to be unreliable  over a wireless signal as more systems get deployed with possible low levels of signal.
NTP uses UDP which would be more reliable in low levels of wireless signal.
While the current code states that not all modems support UDP,
for those new modems that can support NTP there would be a system reliability improvement.
 
So just something to be aware of 
https://github.com/arduino-libraries/NTPClient
if it was something that might be of interest I could submit an update against Modems\DigiXbeeCellularTransparent

SRGDamia1, 2019-09-19:
Right now *none* of the modems support UDP.  Most of are capable of it, but TinyGSM doesn't support it for any of them.

But, yes, NTP over UDP would be better.  It would be the first thing to switch if I write up UDP support in TinyGSM.

neilh10, 2019-09-20:
I have been experimenting with a primitive approach. 
No sucess though. One thing is that behind a fiewall UDP doesn't work - so its not available for WiFi which is typical behind a firewall.  Actually behind a firewall I think requires some sort of tunnel to make it work - and I haven't invested in understanding what that means.
For cellular networks I have been able to get time from the local cell tower sometimes- however its squirly episodic to request and delivered as local time EPOCH 64 bits- so actually need TZ and summertime setting to be reliabel. Buts its close so may work for me..

SRGDamia1, 2019-09-20:
Most of the cellular chips have some sort of command that you can use to get the network time - but the timezone (and string format) of the time you'll get back is pretty grab bag and the documentation isn't always reliable.  Digi's, for example, says that their DT command will return the number of seconds in UTC, but it's actually in local time.

neilh10, 2023-03-17:
I have NTP/UDP over WiFi working on the Wio Terminal.