EnviroDIY/ModularSensors #297: Support for ESP8266 and ESP32

moritz89, 2019-09-20:
The project seams quite mature to be used to measure environmental sensors. I would like to adapt it to be used with ESPs (ESP8266 & ESP32) as these provide a WiFi and Bluetooth interface. Has any work been done in regards to this?

After an initial investigation, includes and functions that have to be addressed include

- pgmspace.h
- Watchdog functionality
- PRINTOUT

SRGDamia1, 2019-09-20:
Right now the ESP8266 is supported as a modem for internet connection, using the "AT" firmware.  I think the ESP32 has nearly identical AT commands when running the Espressif AT firmware, so it could be used in that capacity as well.

To use either as the "main" processor would require some work, like you've seen.  I have both chips on my desk and would love to implement them, but I don't see it happening any time soon just because I don't have time/project money for them.

The issues with pgmspace and PRINTOUT would probably be fairly simple to resolve.  Both are only used for storing strings and printing them out to help debugging.  Most of the "work" would be regex and find-replace to replace the many, many instances of the flash strings.

I think the watch-dog, proper sleep functionality, and real time clock interfacing all present bigger challenges - not insurmountable, but trickier.  The watch-dog issue at least is already "modular-ized."  The clock and sleep support is currently buried inside the logger class (#236 #237).

After all of those are figured out, there would also need to be some hunting down and excluding sensors that aren't supported on the ESP's.  The SDI-12 library, for example, wouldn't run on an ESP so all of those sensors would be off-limits for it.

But, yes, I think it would be a fair amount of work to port over to the Espressif chips, but definitely do able.  I don't think I'm likely to be able to do it anytime soon, but I can review pull requests if you have time!

neilh10, 2023-08-05:
For a base product, seems like the newish CoreS3 might be a reasonable target port. It has a nice mechanical package and display. 
The question can it fit with the a system level of lower cost and as good functionality as the Mayfly 1.1. There are a lot of benefits for a Next Generation processor - including much larger program space.
https://github.com/m5stack/M5CoreS3
https://docs.m5stack.com/en/core/CoreS3

My personal objective is RS485 interface with the ability to connect an LTE modem, as well as analog interfaces. 
The essential value that the Mayfly offers is low power, which translates to small battery and small solar panel.  

The CoreS3 has a power off capability - with a power on from the RTC. That is each sampling run would be a warm powerup.
The CoresS3 has a complex power chip AXP2101 including LiIon battery charging.
The USB-OTG as USB-Device is good for fast program development. 
A uSD for logging data storage.
The ESP32-S3 processor has  3 UARTs - in my system breakout this would be debug, LTE modem, and RS485. A debug UART is required for stability monitoring / testing.
The CoreS3 has an innovative +5V boost scheme off internal battery, all external ports use 5V. Its switched but needs investigating as to are port bits switched completely off.
SDI-12 could be possibly used for a more robustly implementation with DMA - giving a sold timed sampling of the data stream.
For program memory the 16Mbyte flash gets allocated to three program spaces - factory, Upgrade App1, Upgrade App2. That is Over The Air (OTA) appear to be supported - needs investigating.
There is good internal operational ram, but will be lost on power off. There is an external persistent PSRAM

Porting to ModularSensor seems to me the following issues 
1) switching to a new compiler - you don't know issues till its compiler
2) Fitting to the ESP32-S3 API
3) RTC - same issues https://github.com/EnviroDIY/ModularSensors/issues/415 and the API implements a name space
4) Power usage when running - can the clock be wound down to reduce power?

and hardware
5) What size battery(s) and power interface voltage will work?
Probably its going to require development of a DIN Base circuit for implementing similar interfaces to the Mayfly 1.1

neilh10, 2024-06-06:
A new ESP32 board + integrated modem 5G  "Walter" 
and carrier board "Walter feels" - some names - with RS485, RS232, SDI-12 interfaces
is in testing. Apparently then going to be sold through mouser.
 https://www.crowdsupply.com/dptechnics/walter/updates/progress-certifications-and-milestones
From the above a discussion about progress and what it looks like. 
https://www.youtube.com/watch?v=yFCDFZihSak