EnviroDIY/ModularSensors #298: A Modem Subsystem for managing Modem Functions of the Modem Hybrid

neilh10, 2019-09-20:
The modem functions are having some squirly issues - and I've spent lots of time on it and there has been lots of updates …. so this is a my2cents initiative, inline with way the IETF evolves.

From what I can see, I'd like to suggest the modem interface needs to be implemented in a systematic manner as a Modem Subsystem and I have some recommendations for tweaking it to be more reliable.
The Modem Subsystem referes to all aspects of managing the Modem Hybrid (interfaces ports) and power switches.

The various Modem Hybrid(s) are a "communications Subsystem" and effectively implement the "Data Link" Layer 2 and "Physical Layer as defined in the OSI 7 Layer. 
One description is here
https://www.networkworld.com/article/3239677/the-osi-model-explained-how-to-understand-and-remember-the-7-layer-network-model.html 

The Modem Hybrid functions need to be sequenced to work reliably, and I would think its a core function for future connectivity in whatever way that happens.

The different Modem Hybrids are typically complex systems, and may be difficult to test for all stimuli.
The Modem Functions  may experience "equipment" faults and "line" faults - two different types of problems. Equipment faults require the physical device to be replaced, or sometimes if not seated properly to be reseated.
Line "faults" may be intermittent or may, as invalid access parameters  never work. 

I would propose ideally there is a model for the "Modem Subsystem" that manage the stages or states the Modem Hybrid goes through. 
This is largely implemented as an artifact of the design of the amazing  Class TinyGsmClient - but not totally.
class TinyGsmXBee {
public: class GsmClient : public Client
{  friend class TinyGsmXBee;  ..} }
class Client : public Stream {}
class Stream : public Print {}

For a Modem Hybrid function, From power-up to power-down the states are
1a) Powering 
1b) A subset may be "release reset"
1c) - verify/set registers and verify basic Modem Hybrid operation (eg via AT response)
<the following may be repeated>
2) Brought out of sleep (1st cold power up transitions this state) and verify operational. If not valid repower(step 7 then 1)
3) Verify creates/instantiates local wireless link - may have the widest range of problems as wireless as its limit or range is very lossey.
4a) Use active data link in some form. i.e. create TCP links & post,
4b) Management meta data in some form - such as RSSI or local services
5) Shutdown all services (possibly optional)
6) Put to sleep to reduce power. For LTE-M there may be specific requirements to speed wakeup. 

< and repeat to 2) as needed>
7)  Powered down - possibly autonomously, or reset, possibly next step 1a or 1b. 

You could check with your hardware engineer (Shannon?) as to whether there is agreement with the above, or anything missed.

The Meta data such as RSSI or local services only exists at very specific states 

I've been puzzled by the statement in the code DigiXBeeWiFi.cpp:274  - 
    /* The Wifi XBee is unique in that it cannot get signal quality until it */
    /* not only is connected to an access point and has opened a socket, but */
    /* has actually received data on that socket.  There's no command to force */
    /* open the socket, so we do it by sending out one character. */

and then in a creative period lying in bed last night realized what was happening 

 ..metaphorically  speaking the "cart" is being put before the "horse".

[image]

Its true, NO radio gets signal quality before a link of some form or listening in some form to the wireless signal is established.
The issue is that a sensor is created as part of the Modem, 
class DigiXBeeWifi : public DigiXBee : public loggerModem : public Sensor

and that sensor data is attempted to be collected before the Modem is activated. Which currently seems to require the modem be turned on.
For a Logger its very typical to attempt to collect data before the noisy power users such as modems turn on.  
The simple solution is that the Modem Subsystems collects the management data required, stores it locally, and then when a Sensor needs that data, it references the Modem Subsystems last known good copy. ie the RSSI sensors can be implmented as calculated variable [Variable::Variable()  ] who's only operation is to access the value store in Modem Subsystem.
Of course that does mean there is one sample delay from collecting the data to when its included in the data record.   The 1st pass through of the Sensor , it will collect 0.  For a previous project RSSI collection, that turned out to be good enough to have an idea of how the RSSI varied. 

The TinyGsmXbee has been created to manage the complexity of the Modem interface - and doing a great job with that.  I don't believe TinyGsmXbee was created to manage a Modem Subsystem, but that is effectively what it is doing.  I think it would be fairly easy to make it effectively manage the Modem Subsystem - the first step perhaps is the RSSI.

For me personally, any commissioning action (the hugely important installation) the success of steps 1c and 3 should be visible in some form, 
as proposed in https://github.com/EnviroDIY/ModularSensors/pull/290
It would perhaps be part of a Modem Subsystem.

SRGDamia1, 2019-09-20:
Woah.  I'm not sure if it's that you're over-complexifying or just using too much terminology that I'm not familiar with, but I can't follow all of your thoughts.

So for what I do follow:
- Yes, the way we're using the modem as a sensor is doing a little bit of cart-before-horse.  
    - For most of the modules, it's not actually as bad as "getting values before it's active"  The modem is woken up at the very beginning.  
    - For every module except the XBee WiFi, the module will happily report the network signal strength even without having the ability to connect to the network.  Actually, even the WiFi XBee can tell you the signal strength without connecting if you disconnect and run an "active scan."  As far as I'm concerned the modem as a sensor is just measuring the network strength like any other sensor would measure temperature (or like the modem itself measures temperature).
   - If we really want to record the RSSI from the actual transmission of data, you don't need to make a calculated variable for it.  You can just hold the variable in memory.  I'm fine with it always being 1-sample delayed.

As for verifying that the modem is really able to connect to the network at the site of deployment - yes, we could probably increase the number of print-outs about the modem status.  Usually we program everything in the lab, though, and deploy in the field without a laptop.  So the print-outs may or may not ever be seen.  With the GPRSBee, and with the adapter board @s-hicks2 made for the LTE the light flash pattern is much more helpful.  The reason that I didn't like waiting for an IP in the setup as you did in #290 is that it can take a while to get that IP assigned - you really should have a waitForNetwork() step in there - and I'd rather off-load that wait until the internet connection is actually needed.

neilh10, 2019-09-21:
Well what bit do you think is too much terminology?.  

I tried to break it down into all the steps that are needed, to figure out why its being so squirly.

FYI, from creating a quiet electrical environment for analog sensors (good analog management),  and reducing dynamic battery management power demands its good practice to run the RF after the sensing.

SRGDamia1, 2019-09-21:
I'm sorry, some how putting things in terms of states and subsystems and models makes me feel panicked at the complexity, even if it's correct terminology for what's already in place.

neilh10, 2019-09-21:
Well I guess I'm describing something (possibly to myself ) for what I've seen work and what I may need to tweak. 
I've also been in environments with a team of different expertise, "a scrum", and a descriptive/story model is sometimes used.   With an electronics background I'm coming at it from the hardware up and having worked with embedded designs I've seen the value of state based protocols. That is what it takes to manage the hardware.  The Class structure is great (though sends me looking at manuals sometimes to see if I can decode why its working), but its top down on the hardware.  The hardware up approach only works to the point of managing the communications subsystem. 
For my initial testing after integrating to 0,23.11 I with WiFi Digi S6b I was getting very poor results. Now a day later it seems a lot better, and I'm not sure why.   
I'm also capturing the UART/Digi trace (with a saleae logic analyzer) to have a record of what works, and possibly decode what fails. Communication subsystems are always complex.
I guess I should probably take this off to my fork as a reference.

SRGDamia1, 2019-09-24:
I've been thinking more about this, and I think you're right about the order things ought to go in.  It will take some refactoring, but I think it might actually simplify the implementation.

aufdenkampe, 2019-09-24:
@SRGDamia1 & @neilh10, I very much appreciate the dialog you have, especially coming from your different perspectives/expertise in object oriented programming and embedded systems engineering respectively. I'm often in the position on not understanding a lot of the terminology you each use, and getting overwhelmed by how complicated it all seems. At least that's often my first reaction. The power of these interactions, and also your willingness to have them in public, is that each of us eventually steps back, re-parses out the info for ourselves in more digestible chunks, and ponders the ideas, which leads to innovation. This has led to ModularSensors and associated software/hardware solutions to be the impressively capable system that it has become. Thank you both!

One idea that I really like from @neilh10's suggestion is to keep the radio off during the sensing cycle, and to report radio variables from the previous sensing cycle (which is something we do for the Trigger function that I have used, and probably a few other things). I like that idea because:
- It minimizes potential RF interference to analog sensors, which I've recently experienced with tipping bucket applications (maybe it wasn't from the radio, but maybe it was).
- It minimizes power usage, because radios are such a power hog.

There may be other good ideas that I haven't yet fully understood or digested.

neilh10, 2019-09-24:
Hi @SRGDamia1 and @aufdenkampe - I really appreciate  the great Class structure and encapsulating the complexity behind it, and the fantastic set of sensors that are interworked.

In my terminology  "Modem Management Subsystem" I'm phrasing a generic way of discussing it, which if appropriate would be a Class structure. 
The reference to Finite State Machines (FSM) is that it can be a very simple coding tool - but pushes the work into design phase,
eg  gsmModem.commandMode()  is a simple two state FSM. However it depends on other states that are not represented. 

 The FSM also defines where the responsibility lies - eg if a Modem locks up (doesn't behave as expected) - then it should be reset in the state trying to bring it up.
FSM are definitely a "deep topic" and personally I find looking for UML FSM more useful.
(eg wikipedia entry on FSM I don't understand) 

https://en.wikipedia.org/wiki/UML_state_machine
https://online.visual-paradigm.com/diagrams/tutorials/state-machine-diagram-tutorial/

In the terminology of the day - there is an object that is equipment "Modem Hybrid" that has a number of states (OFF-state,xx-state,yy-state,ON-InService) and there is a Communication Services Layer  (~GsmClient) that use the equipment when it is IN-Service states. 

So in the Modem Management System - it manages the equipment Modem Hybrid.  It has unique states from Inactive/Power Off, Power On-Searching,  to InService. It would probably have a state=EquipmentFault which is some response  (or no response) happened that indicates no-operation or undesigned for. 

The Communication Services are only applicable when the equipment is in the IN-Service states.
Also, looking to the future, only one Communication Service/Socket (a TCP or UDP) is supported at any one time currently .  However  the hardware Modem Hybrid can support 6 sockets which may be useful in the future. The service layer decouple this, and either accepts a request or rejects it if there aren't available resources (eg power on, or in the single Socket its not available.) 

I do appreciate the open source side, and that there are real deliverables that you have, so only take what is useful.

On the radio side - I've seen some horrible breakthrough from noisy RF sources. In urban settings, and the test bench,  these can be very uncontrollable - like taxis or GSM phone burst. 
On the test bench I've also tried to seal up against electrical noise and its very hard. 
IMHO the simpler solution is to keep sensing and communications to different time domains. 
That also supports the dynamic powering if there is a sensor that is a heavy user.

neilh10, 2019-09-26:
FYI - using the LTE-M ver 11413 with  code based on 0.23.15 it worked excellently. Yahhoooo.!! 
It is posting  to data.enviroDIY.org and the time update worked. 
Really great to see.!!! Thankyou thankyou. 
The current readings were also excellent, though I need to do some further characterization, though I may not have the shutdown configured so looking at that.

SRGDamia1, 2019-09-26:
Can you look at the restructuring on the modemLast branch?

neilh10, 2019-09-27:
Are the changes addressing one issue, or covering a number of issues. 
Hard to review without identifying the scope - thanks for putting them forward, not sure how much help I can be.

So I think 1) sampling ModeMeta data after comms:
the core seems to be to reasonable to do the sampling after the modem has run.
loggerBase.cpp:
publishDataToRemotes();
_logModem->updateModemMetadata();

2) ?? is this addressing better reliability
 are these part of 1), perhaps a necessary band aid to make it work, or new functions?
LoggerModemMacros.h
bool specificModem::modemWake(void)
….
        /* Check the status pin and wake bits before running wake function */                    \
        /* Don't want to accidently pulse an already on modem to off */                          \
        /* NOTE:  It's possible that the status pin is on, but the modem is actually */          \
        /* mid-shutdown.  In that case, we'll mistakenly skip re-waking it. */                   \
        if (_statusPin >= 0 && digitalRead(_statusPin) == _statusLevel && !_alwaysRunWake)

I find hard to read and decipher (maybe late at night for me)
(_statusPin >= 0 && digitalRead(_statusPin) == _statusLevel && !_alwaysRunWake)
assuming it means
((_statusPin >= 0 && digitalRead(_statusPin)) == (_statusLevel && !_alwaysRunWake))

I guess it seems very complicated, but I would need to try and walk the calling chain to understand all the options and see if there were reasonable suggestions, and probably even then need to run it and see if it hung together. 
In the end what counts - does it work … perfectly... or at least as reliably as before. Oops forgotten there is a real world.!!

neilh10, 2019-09-29:
Hey got to do some testing on 0.23.15 over Xbee3 LTE-M ~ and it was working pretty well. I haven't got any Xbee3 sensor configured so not trying to run that.  Really want to say thankyou thankyou.

I kinda feel mean making comments about an alternate structure ..... working code in hand is always fantastic  ... :)
So the new code, on an eyeball analysis  for updateMetaModem looks good and fair amount of code changes.   I'm assuming testing across a few devices  .... which is a lot of the work. 

My 2cs thought would be to say is it possible to keep the same calls in LoggerBase.cpp
_logModem->connectInternet())   .. its doing the service request "requestConnectTgt"and you don't want it to know how to do anything with the underlying layers. It either works or fails.
Maybe that pushes too much into LoggerModemMacros.h - I must admit while I follow the code in LoggerModemMacros.h looks a pretty amazing sequence to map to the various modems via macros.

I still have an idea that there would be a benefit in having the ModemState visible- but it might need a whiteboard to sketch out the idea. 
Right now the ModemState information is implicit in the structuer of the methods. 
By making it explicit variable "enum ModemSate"  (1a-7 above) it would make it easier to deal with an unexpected event. If modemWake() encounters unexpected error, it would just change the ModemState=RESET_STATE and the state machine would deal with it as if it was powerup. Right now the modemWake() repeats in code the startup.    

Oops I'm getting a signal my times up, gotta go party ....  as I said whatever works for you to have working code. :) Thankyou thankyou.

neilh10, 2023-09-21:
The modem subsystem with TinyGSM is effectively performing this function.

closed 2023-09-21 by neilh10