EnviroDIY/ModularSensors #301: SARA-U201 only sending data every other logging interval

dnowacki-usgs, 2019-12-02:
I'm using a u-blox SARA-U201 with a Mayfly v0.5b and ModularSensors 0.23.16. Using a modified version of menu_a_la_carte.ino, I find that the modem is only activated every other logging interval. This is with relatively short intervals (2 and 5 minutes). The Mayfly goes through boot-up and NIST checking without issue. At the first interval, status pin 19 is low, indicating OFF, and it bails on doing any modem stuff before saving to the SD card. At the second interval, it does a hard reset on the modem, and successfully connects. At the third interval, it gets the same status pin low like at the first interval. I'm using the constructor from this [wiki page](https://github.com/EnviroDIY/ModularSensors/wiki/Sodaq-uBee). What's going on?

```
This logger has a variable array with 12 variables, of which 11 come from 3 sensors and 1 are calculated.
Sampling feature UUID is: 12345678-abcd-1234-ef00-1234567890ab
Logger portion of setup finished.

Power to unspecified modem is not controlled by this library. <--LoggerModem
Running wake function for unspecified modem <--LoggerModem
unspecified modem should be awake. <--LoggerModem
Setting up the modem ... <--LoggerModem
Modem was already awake and should be ready for setup. <--LoggerModem
Running modem's begin function ... <--LoggerModem
... Complete!  It's a u-blox SARA-U201 <--LoggerModem
u-blox SARA-U201 warms up in 0 ms, indicates status in 100 ms, is responsive to AT commands in less than 6000 ms, and takes up to 15000 ms to close connections and shut down. <--LoggerModem
Leaving modem on after setup ... <--LoggerModem
Attempting to connect to the internet and synchronize RTC with NIST
Response Byte 0 : � = 225 = 11100001 <--LoggerModem
Response Byte 1 : � = 143 = 10001111 <--LoggerModem
Response Byte 2 : � = 240 = 11110000 <--LoggerModem
Response Byte 3 : � = 192 = 11000000 <--LoggerModem
Seconds from Jan 1, 1900 returned by NIST (UTC): 3784306880 = 11100001100011111111000011000000 <--LoggerModem
Unix Timestamp returned by NIST (UTC): 1575318080 <--LoggerModem
Clock set!
Setting up sensors...
Setting up file on SD card
Data will be saved as XXXXX_2019-12-02.csv
Putting modem to sleep
Turning u-blox SARA-U201 off. <--LoggerModem
Running given sleep function for u-blox SARA-U201 <--LoggerModem
Total modem active time (s): 26.151 <--LoggerModem
Power to u-blox SARA-U201 is not controlled by this library. <--LoggerModem
Putting processor to sleep

------------------------------------------
Power to u-blox SARA-U201 is not controlled by this library. <--LoggerModem
Skipping u-blox SARA-U201 in sensor power up! <--LoggerModem
Running wake function for u-blox SARA-U201 <--LoggerModem
u-blox SARA-U201 should be awake. <--LoggerModem
It's been 622 ms, and status pin 19 on u-blox SARA-U201 is 0 indicating it is off.  Will not continue to attempt communication! <--LoggerModem
u-blox SARA-U201 is not expected to return measurement results! <--LoggerModem
PRIOR modem active time: 26.151 <--LoggerModem
PRIOR modem powered time: -9999.000 <--LoggerModem
Skipping u-blox SARA-U201 in sensor power down! <--LoggerModem

 \/---- Line Saved to SD Card ----\/
2019-12-02 20:22:00,2,11103,4.791,22.50,-9999,-9999,-9999,-9999,-9999,-9999.0,26.151,-9999.000

Turning u-blox SARA-U201 off. <--LoggerModem
Status pin 19 on u-blox SARA-U201 is 0 indicating it is already off!  Will not run sleep function. <--LoggerModem
Power to u-blox SARA-U201 is not controlled by this library. <--LoggerModem
------------------------------------------

------------------------------------------
Power to u-blox SARA-U201 is not controlled by this library. <--LoggerModem
Skipping u-blox SARA-U201 in sensor power up! <--LoggerModem
Running wake function for u-blox SARA-U201 <--LoggerModem
u-blox SARA-U201 should be awake. <--LoggerModem
It's been 618 ms, and status pin 19 on u-blox SARA-U201 is 0 indicating it is off.  Attempting a hard reset on the modem! <--LoggerModem
Doing a hard reset on the modem! <--LoggerModem
It's been 4915 ms, and u-blox SARA-U201 is now responding to AT commands! <--LoggerModem
It's been 5165 ms, and u-blox SARA-U201 is now responding to AT commands! <--LoggerModem
It's been 5415 ms, and u-blox SARA-U201 is now responding to AT commands! <--LoggerModem
It's been 5665 ms, and u-blox SARA-U201 is now responding to AT commands! <--LoggerModem
It's been 5915 ms, and u-blox SARA-U201 is now responding to AT commands! <--LoggerModem
It's been 6166 ms, and u-blox SARA-U201 is now responding to AT commands! <--LoggerModem
It's been 6416 ms, and u-blox SARA-U201 is now responding to AT commands! <--LoggerModem
It's been 6666 ms, and u-blox SARA-U201 is now responding to AT commands! <--LoggerModem
It's been 6916 ms, and u-blox SARA-U201 is now responding to AT commands! <--LoggerModem
It's been 7166 ms, and u-blox SARA-U201 is now responding to AT commands! <--LoggerModem
[... 44 lines cut ...]
Turning u-blox SARA-U201 off. <--LoggerModem
Running given sleep function for u-blox SARA-U201 <--LoggerModem
Total modem active time (s): 34.701 <--LoggerModem
Power to u-blox SARA-U201 is not controlled by this library. <--LoggerModem
------------------------------------------

------------------------------------------
Power to u-blox SARA-U201 is not controlled by this library. <--LoggerModem
Skipping u-blox SARA-U201 in sensor power up! <--LoggerModem
Running wake function for u-blox SARA-U201 <--LoggerModem
u-blox SARA-U201 should be awake. <--LoggerModem
It's been 618 ms, and status pin 19 on u-blox SARA-U201 is 0 indicating it is off.  Will not continue to attempt communication! <--LoggerModem
u-blox SARA-U201 is not expected to return measurement results! <--LoggerModem
PRIOR modem active time: 34.701 <--LoggerModem
PRIOR modem powered time: -9999.000 <--LoggerModem
Skipping u-blox SARA-U201 in sensor power down! <--LoggerModem

 \/---- Line Saved to SD Card ----\/
2019-12-02 20:26:00,4,11103,4.776,22.75,-9999,-9999,-9999,-9999,-9999,-9999.0,34.701,-9999.000

Turning u-blox SARA-U201 off. <--LoggerModem
Status pin 19 on u-blox SARA-U201 is 0 indicating it is already off!  Will not run sleep function. <--LoggerModem
Power to u-blox SARA-U201 is not controlled by this library. <--LoggerModem
------------------------------------------
```

SRGDamia1, 2019-12-03:
Are you using a Sodaq UBee with a SARA U201 chip and a Mayfly?  Are you sure you have the correct pins assigned?  Unless you specifically chose not to power the modem and manually connected the reset, I suspect you do not.

For a Mayfly with a UBee-201 the pins should be:
```cpp
// Sodaq uBee U201 with Mayfly
const int8_t modemVccPin = 23;      // MCU pin controlling modem power (UBee pin 9/Mayfly pin 23 is connected to a MIC94040 which controls the power supply to the U201)
const int8_t modemStatusPin = 19;   // MCU pin used to read modem status (UBee pin 12/Mayfly pin 19 is connected to U201's v_int via another MIC94040)
const int8_t modemResetPin = -1;    // MCU pin connected to modem reset pin (UBee pin 5 is connected to reset_N on the U201, but this pin is not connected to the Mayfly mcu)
const int8_t modemSleepRqPin = 20;  // MCU pin used for modem sleep/wake request (UBee pin 16/Mayfly pin 20 is directly connected to the U201's pwr_on pin - this (and only this) pin on the U201 is 5.5V tolerant)
const int8_t modemLEDPin = redLED;  // MCU pin connected an LED to show modem status (Optional, but I like to use it)
```

SRGDamia1, 2019-12-03:
This is based on the schema I have for the UBee dated 01/25/18, rev 1.  If there have been changes since then, I do not have an updated schema

dnowacki-usgs, 2019-12-03:
Thank you, it was the wrong pins. I searched quite a while looking for the pins for the uBee but couldn't find it on the GitHub wiki or elsewhere... Where is the best place to find the pins for other modems?

SRGDamia1, 2019-12-03:
I should add it to the wiki, shouldn't I?

For the most part, you have to search for the schematic of whatever you have and figure it out yourself right now.

closed 2019-12-03 by SRGDamia1

SRGDamia1, 2019-12-04:
I've added pin numbers to the wiki for the UBee with ModularSensors:  https://github.com/EnviroDIY/ModularSensors/wiki/Sodaq-uBee

You could have figured out the right pins by looking at the [Mayfly schematic](http://www.envirodiy.org/wp-content/uploads/mayfly_v0p5b_schematic.png) and the [UBee schematic](https://support.sodaq.com/Shields_and_Bees/sodaq_ubee.pdf), but I definitely understand why you would have been confused looking at the two schematics.  My copy of the UBee schematic is heavily annotated and I just added a few more scribbles to it today making sure I had everything right to put into the wiki.
[image]