EnviroDIY/ModularSensors #328: Building menu_a_la_carte fails

neilh10, 2020-12-03:
I'm trying to do as a Newbie for some basic starting, and its failing to work.
 
For a new system, I always try and build an example - just to test the toolchain and setup and learn the basics.
The instructions point to menu_a_la_carte to do this.

I'm going to 
https://github.com/EnviroDIY/ModularSensors/wiki/Getting-Started
with PlatformiIO/VSC installed and working. 
Then for 
https://github.com/EnviroDIY/ModularSensors/tree/master/examples/menu_a_la_carte
From PIO I use File->Open Folder then navigate to ModularSensors\examples\menu_a_la_carte and open it. 
It opens.
In PIO I click Build, 
then I get a failure coming up...

Error: Nothing to build. Please put your source code files to 'D:\usr\a\Documents\Arduino\envtmp\ModularSensors\examples\menu_a_la_carte\src' folder

-----------

Practically speaking I can make this build, but its not doing it as per the instructions.

I'm looking to write up this in a blog - and reference the instructions.
Perhaps I've missed some special setup.

SRGDamia1, 2020-12-03:
Thank you for the call-out.  I was assuming for the examples that someone would create a brand new project in pio and copy the code in the example into the main.cpp in the src folder of the new project and the ini over the new project ini.  Obviously, that's not clear.  You could also create a new project, have MS in the lib_deps section and then build with the src_dir = .pio/libdeps/[env name]/EnviroDIY_ModularSensors_####/examples/menu_a_la_carte, but I think digging into that deep a folder structure is very annoying.

The examples folder structure (ie, a folder with the name of the example and then inside an ino of identical name) is for the Arduino IDE.  I think it's an awkward convention.

neilh10, 2020-12-04:
I really appreciate the broad overview you cover and I'm still getting into the all the new docs - which no doubt helps browse the libs

From a newbie ease of use Arduino style  example (or old hand doing a quick test)  - a folder/project that can be opened and  does a full  compile/download gives a good sense the tool chain works and learning what are the right buttons and sequence to do things in. 
So from a clean machine

Install PIO/VSC
(download ModularSesnors.git.zip or install git, and git clone https://github.com/EnviroDIY/ModularSensors)

Then a PIO/VSC file->Open Folder
PIO Build and success link - which implies all libs can be pulled in.
Physically connect Mayfly to USB, 
Then a build download 
PIO TTY and check all the TTY response works etc.

It maybe just needs to be one menu_a_la_carte

I think its just pushing menu_a_la_carte.ino into src/menu_a_la_carte.ino and possibly adjusting platformIo.ini

SRGDamia1, 2020-12-04:
If you're using the Arduino IDE, you can also get right to the examples for a library by going to File > Examples and scrolling down to the custom library name.

We do still teach the Arduino IDE in all our workshops.  I've made *zero* headway in convincing others here to use VSCode or Atom or anything else.

neilh10, 2021-02-10:
Will check out per examples.

closed 2021-02-10 by neilh10