EnviroDIY/ModularSensors #332: Broken links to Getting Started page

mbarneytu, 2020-12-09:
There are two places where the link to the Getting Started page is broken. One is in the first body paragraph [here](https://envirodiy.github.io/ModularSensors/index.html), and the second is in the first paragraph on the [Readme page ](https://github.com/EnviroDIY/ModularSensors)in the main repo.

SRGDamia1, 2020-12-09:
Ouch.  Sorry.  Neil had also pointed that out and I'd fixed it on develop but not pushed to master (which is where you were looking).

I pulled develop through with most of the fixes.  Let me know if you've found any others.

Also, the documentation at https://envirodiy.github.io/ModularSensors/index.html has been heavily re-styled to look much more like the EnviroDIY webpage.

SRGDamia1, 2020-12-09:
The doc's built on the github.io page are updated whenever Travis finishes a job, so it will be several hours before they reflect the change.

closed 2020-12-09 by SRGDamia1