EnviroDIY/ModularSensors #334: Migrate away from Travis for CI

SRGDamia1, 2020-12-09:
Travis doesn't have unlimited free minutes for open source any more and it's been really, really backed up recently.

SRGDamia1, 2020-12-15:
Ok, while rather more difficult than my reading suggested it might be, I've migrated all of our workflows to Github Actions.  Now to see the results, you'll be able to click the ["Actions" link ](https://github.com/EnviroDIY/ModularSensors/actions) in the repo rather than needing to go off-site to Travis.

There are currently 5 workflows:
- `verify_library`_json runs on all pushes or pull requests to all branches and simply verifies that library.json is valid json
- `build_examples` runs on all pushes or pull requests to all branches and builds the 10 examples in several different environments to look for compiler errors
- `build_documentation` runs on pushes to the master branch or on the creation of a new release and runs doxygen and m.css and publishes the results to github.io.
- `prepare_release` runs on pushes to the master branch when the `version` file changes.  It checks the current version in the version file, downloads and zips up the library and all its dependencies, fetches the text for the release from the chagnelog file, creates a new release with the version tag, uploads the zip file to the release, and finally publishes the library to the platformio registry.  Whew.
- `changelog_reminder` automatically comments on all pull requests reminding you to update the changelog file, if it hasn't been modified.

SRGDamia1, 2020-12-15:
At the present, Github actions seem to be running faster than Travis.  Though, ironically, Travis was less than an hour behind for me yesterday.

closed 2020-12-15 by SRGDamia1

aufdenkampe, 2020-12-15:
This is fantastic @SRGDamia1. Thanks for researching this, and then implementing it.
It's also great to know that an official release will be automatically triggered when the `version` is updated on the master branch.