EnviroDIY/ModularSensors #335: Keep doxgen generated documentation in sync with develop?

SRGDamia1, 2020-12-09:
Opinions?  @aufdenkampe @neilh10

Right now the documentation is re-built with the master branch only.  I was thinking it might be better to keep it with the develop branch instead - that is, with what's coming.

neilh10, 2020-12-09:
I would say target for master branch only. 
The reason for develop would be to check it and provide preview.  So could be done at the end of the develop cycle if needed.

neilh10, 2020-12-09:
Its amazing to see how the documentation is created ~ really putting ModularSensors out front.  
I've used docs before to pickup on an API, and appreciated it with out any idea how it was created.  
For me though at this point the code through PIO is what I'm using, so I may not be the best person to comment.  I guess its really who is looking at the docs and might want to commenting on (develop)
Linking people across a team is rally valuable - would it be worth seeing if Beth wants to comment on them?

aufdenkampe, 2020-12-10:
@SRGDamia1, that is a great question. I'm leaning toward keeping the doxygen docs in line with `master`, as that is what we want most people to be using. If someone is contributing to `develop`, then they should be savvy enough to see how things might be different between the two. 

Of course, the ideal thing would be to have both, by using subfolders in the `gh-pages` branch, one for the latest release on Master (or all releases?) and one for develop. We did that for the dbWrench documentation for ODM2, to some extent, and it worked well.

neilh10, 2020-12-11:
Another doc issue - these don't have landings
https://envirodiy.github.io/ModularSensors/for_developers.html
https://envirodiy.github.io/ModularSensors/library_terminology.html

from 
https://envirodiy.github.io/ModularSensors/#mainpage_data_receivers "tips here"   "terminology

SRGDamia1, 2020-12-15:
I think I've fixed all the links.

The docs will stay in sync with master.

closed 2020-12-15 by SRGDamia1

aufdenkampe, 2020-12-15:
@SRGDamia1, Thanks! And thanks for issuing the official release!

aufdenkampe, 2020-12-15:
BTW, how did you get "github-actions" to issue the release? Did you trigger that?

SRGDamia1, 2020-12-15:
This workflow does it:  https://github.com/EnviroDIY/ModularSensors/blob/master/.github/workflows/prepare_release.yaml

It will be run automatically when the `version` file changes on the master branch.  It can also be triggered manually via the [action page](https://github.com/EnviroDIY/ModularSensors/actions?query=workflow%3A%22Prepare+a+new+release+and+add+release+assets%22).

See https://github.com/EnviroDIY/ModularSensors/issues/334#issuecomment-745430050