EnviroDIY/ModularSensors #346: SDI12 new feature can result in watchdog biting

neilh10, 2021-01-05:
For the new SDI-12 features introduced in 0.27.0 they can result in some sort of loop that causes the watchdog to fire.

I haven't located exactly where this happens in the SDI-12 code yet, but it happens for me when it is trying to poll an Insitu LT500 sensor, and happen after some time.
I think these features are in https://github.com/EnviroDIY/ModularSensors/pull/333  SDI12Sensors.cpp

The Insitu LT500 polling worked for me in the 0.25.0 release.  In 0.27.5 the Insitu LT500 is not respond to the new queries - and still investigating. So this is a headsup and disccussion element in case anybody else is seeing this.

neilh10, 2021-01-06:
Well I'm zeroing in on the Watchdog biting - it does read the LT500/SDI12, then it never leaves - Woof.!
Possible one complexity is that I have a Keller RS485 as well, but wasn't a problem in 0.25.0
The problem is I'm not really understanding the intent of the new functionality for SDI12Sensors.cpp 
I can monitor the SDI12 with a logic analyzer and see that it works ~ cmd and response. 
When I turn on   -DMS_SDI12SENSORS_DEBUG  it messes with the timings and the LT500 responds to "1!" but not to anything more complex like "1l!" or "1C!" or  "1D0!"

Using the 0.25.0 version of SDI12Sensors::startSingleMeasurement(void) { } in a define MS_SDI12_BACK_COMPATIBLE, it works and watchdog OK. T
he SDI-12 cmds look good and with sometime are
4] sec powerup
4.493] 1! rsp 1
1I! rsp 113IN-SITU LT500 306 0000525528
4.9] power off
73.5] power up
74.075]1! rsp 1
74.140]1C! rsp 100103\r\n
74.165]1D0! rsp 1+0.10599+15.3675+0.24472\r\n
75.048]1! rsp 1\r\n
75.120]1C!  rsp 100103\r\n
75.746]1D0! rsp 1+0.10582+15.3705+0.24434\r\n
84] Power off

With latest SDI12sensors something goes into the loop between the 2ndPoll, possibly kicking in with possibly SDI12sensors.cpp#367 while (_SDI12Internal.available()) 

the line (without the timeings)
\x7F\0    - power up
1!  rsp 1\n\r
1!  rsp 1\n\r
1I! rsp 113IN-SITU LT500 306 0000525528\n\r
Power down

\x7F\0  - powerup
1!   rsp 1\n\r
1C!  rsp 100103\r\n
1D0!  rsp 1+0.10633+15.4759+0.24551\r\n
(then loop with watchdog kicks in)

With debug     -DMS_SENSORBASE_DEBUG  -DMS_VARIABLEARRAY_DEBUG
this is what it looks like
3 --->> Waking InsituTrollSdi12 at SDI12-1_Pin7 ... <--VariableArray
Waking InsituTrollSdi12 at SDI12-1_Pin7 <--SensorBase
   ... wake up uccess. <<--- 3 <--VariableArray
3 . 1 --->> Starting reading 1 on InsituTrollSdi12 at SDI12-1_Pin7 ... <--VariableArray
Starting measurement on InsituTrollSdi12 at SDI12-1_Pin7 <--SensorBase
   ... set up succeeded. <<--- 3 . 1 <--VariableArray
5 --->> Waking KellerAcculevel at modbus_0x01 ... <--VariableArray
Waking KellerAcculevel at modbus_0x01 <--SensorBase
   ... wake up uccess. <<--- 5 <--VariableArray
3 . 1 --->> Collected result of reading 1 from InsituTrollSdi12 at SDI12-1_Pin7 ... <--VariableArray
Putting 1.00 in result array for variable 0 from InsituTrollSdi12 at SDI12-1_Pin7 <--SensorBase
 WATCHDOG ISR barksUntilReset 51 <--WatchDogAVR
 WATCHDOG ISR barksUntilReset 50 <--WatchDogAVR
 WATCHDOG ISR barksUntilReset 49 <--WatchDogAVR
 WATCHDOG ISR barksUntilReset 48 <--WatchDogAVR
 WATCHDOG ISR barksUntilReset 47 <--WatchDogAVR
 WATCHDOG ISR barksUntilReset 46 <--WatchDogAVR
 WATCHDOG ISR barksUntilReset 45 <--WatchDogAVR
 WATCHDOG ISR barksUntilReset 44 <--WatchDogAVR
.... till Woof

Trying some other variants and also trying to figure out why the timing is upset with-DMS_SDI12SENSORS_DEBUG
 @SRGDamia1  I wonder if you have any insights.?

SRGDamia1, 2021-01-06:
I'm sorry, I haven't looked at this yet.

The two changes to the SDI-12 class were to

- Allow using an M! command to initiate a measurement instead of the C! command.  
- Continue to poll the sensor with D#! commands if the expected number of values don't return until either they all arrive or nothing else arrives.

I suspect the latter is the issue.

When the SDI-12 sensors send data in response to a D! /D0! command, the size of the response is pretty limited (81ish characters).  So to get more values, you poll again with increasing D#'s.  Previously, none of the sensors had more than 3 values and they all fit in the first call, but with the RDO I needed more values so I stated polling 0-9 until either the expected number or 0 values are returned.  There's probably something not quite right in the logic there that is causing that to endlessly loop.

The M!/C! is the `MS_SDI12_NON_CONCURRENT` option.  Basically, with a concurrent measurement (C!), you're supposed to be able to tell the sensor to take a measurement, and then it should proceed to take the measurement and wait until you ask it for data.  While it's taking the concurrent measurement, it should ignore everything on the line except an explicit command to its own address.  For fairly old sensors, or ones where the manufacturer screwed up the implementation (Insitu RDO..), the other option is a measurement with the M! command.  With the M! command, you must wait until the measurement on that sensor is finished and the sensor sends back a service request (just an !).  If there's any other command on the line before the sensor finishes, it aborts the measurement.  So it's not ideal at all because if you have multiple sensors you have go one at a time start - wait - finish rather than starting all three at once and then asking for data when they ought to have finished.  And, because of the way ModSen is structured, you can't go on to even non SDI-12 sensors in the interim.  Unless you're super limited with pins and you just have to have two SDI-12 sensors on the same pin and they can't play nice, there's no reason to use this.  That's why you have to explicitly define `MS_SDI12_NON_CONCURRENT` to use this and it then applies to everything.

SRGDamia1, 2021-01-06:
Ok. first issue is with https://github.com/EnviroDIY/ModularSensors/blob/d166bbd014f57af74ba5fdf154daf58813b02dbf/src/sensors/SDI12Sensors.cpp#L364 where I forgot to eat the repeated address if the debugging is not on.  That's why you're getting the first "result" is the address.

SRGDamia1, 2021-01-06:
Do you have the latest SDI-12 library (from Dec 1)?  I added an extra 10ms wake delay by default, maybe that's screwing it up.  You can try getting rid of it by either modifying the send command function in the SDI-12 library or adding the optional wake delay of 0 for all of the sendCommand functions (`sendCommand(blah, 0);`)

neilh10, 2021-01-07:
Oh interesting thanks for the explanation.
I haven't dug in to SDI-12 as it worked pretty well for me up to 0.25.0   
For 0.25.0 addSingleMeasurementResult() no debug it works with the LT500
For 0.27.5 the poll works, but the processing causes the problem, but for me I can't debug the structure,  as the debug statements seem to be changing the timing  - at least LT500 doesn't respond. I'm left with trying to remove debug statements to find out why not. 
The M command has a compile issue - doesn't look it has ever been tested. 

 Would you be ok with a compile switch for the "single 81 char" simple version the way it was for 0.25.0 while working through the options for the more elaborate polling.?  

I always find being able to trace compatibility is a benefit. I have a device in the field that I need to support
I'd be happy to submit a PR with above switch.  

For the 0.27.5 library.json  spec for "1486" pulls in something slightly older than https://github.com/EnviroDIY/Arduino-SDI-12.git" but is pretty minor

(1486 has)  uint8_t _dataPin;     (newer version has) int8_t _dataPin;

      "name": "SDI-12",
      "owner": "envirodiy",
      "library id": "1486",
      "version": "~2.1.2",
      "url": "https://github.com/EnviroDIY/Arduino-SDI-12.git",

For (0.27.5)  .pio\libdeps\mayfly\SDI-12\SDI-12.cpp  [which has worked with the 0.25.0 addSingleMeasurementResult() }] on DIsk I have
  "name": "SDI-12",
  "version": "2.1.2",
    "url": "https://github.com/EnviroDIY/Arduino-SDI-12.git"

neilh10, 2021-01-07:
With bool SDI12Sensors::getResults(void) {
When its parsing the response from LT500 
1D0!  rsp  "1+0.10566+16.5545+0.24396"
 it is not parsing the "+"   which is then causing the Woof

I put in a 
            delay(100);  // was 10 ~ 1 character ~ 7.5ms
            STANDARD_SERIAL_OUTPUT.write(c);

3 . 1 --->> Collected result of reading 1 from InsituTrollSdi12 at SDI12-1_Pin7 ... <--VariableArray
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++ WATCHDOG ISR barksUntilReset 51 <--WatchDogAVR
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++ WATCHDOG ISR barksUntilReset 50 <--WatchDogAVR
++++++++++++++++

neilh10, 2021-01-07:
For an LT500 transactions
\x7F\01!1
1!1
\01I!113IN-SITU LT500 306 0000525528
\x7F\01!1
\01C!100103
\01D0!1+0.10555+16.6187+0.24371
\01!1
\01C!100103
\01D0!1+0.10563+16.6166+0.24390

this code parse it, by using '+'

`    while (resultsReceived < _numReturnedValues && cmd_number <= 9) {
        bool   gotResults     = false;
        String getDataCommand = "";
        getDataCommand += _SDI12address;
        // SDI-12 command to get data [address][D][dataOption][!]
        getDataCommand += "D";
        getDataCommand += cmd_number;
        getDataCommand += "!";
        MS_DEEP_DBG(F("    >>>"), getDataCommand);
        _SDI12Internal.sendCommand(getDataCommand);
        delay(30);  // It just needs this little delay

        uint32_t start = millis();
        while (_SDI12Internal.available() < 3 && (millis() - start) < 1500) {}
        //MS_DBG(F("  Receiving results from"), getSensorNameAndLocation());
        sdiNextChar = static_cast<char>(_SDI12Internal.read());
        MS_DEEP_DBG(F("    <<<"),sdiNextChar);
        // ^^ ignore the repeated SDI12 address

        start = millis();
        while (0 < _SDI12Internal.available()) {
            int c = _SDI12Internal.peek();
            if (c == '-' || (c >= '0' && c <= '9') || c == '.'**|| c == '+'**) {
                float result = _SDI12Internal.parseFloat();
                // The SDI-12 library should return -9999 on timeout
                if (result == -9999 || isnan(result)) result = -9999;
                MS_DEEP_DBG(F("    <<<"), String(result, 10));
                verifyAndAddMeasurementResult(resultsReceived, result);
                if (result != -9999) {
                    gotResults = true;
                    resultsReceived++;
                }
            } else if (c >= 0 && c != '\r' && c != '\n') {
                MS_DEEP_DBG(F("    <<<"), static_cast<char>(_SDI12Internal.read()));
            } else {  // no point -1's and new lines to debugging port
                _SDI12Internal.read();
            }
            **delay(100);  // was 10 ~ 1 character ~ 7.5ms
            STANDARD_SERIAL_OUTPUT.write(c);
            if ((millis() - start) > 10000) {
                PRINTOUT("SDI parse timeout");
                PRINTOUT(_SDI12Internal.readStringUntil('\n'));
                break;
            }**
        }
        if (!gotResults) {
            MS_DBG(F("  No results received, will not continue requests!"));
            break;  // don't do another loop if we got nothing
        }
        MS_DBG(F("  Total Results Received: "), resultsReceived,
               F(", Remaining: "), _numReturnedValues - resultsReceived);
        cmd_number++;
    }`

the output with limited debug was
3 . 1 --->> Starting reading 1 on InsituTrollSdi12 at SDI12-1_Pin7 ... <--VariableArray
Starting measurement on InsituTrollSdi12 at SDI12-1_Pin7 <--SensorBase
   ... set up succeeded. <<--- 3 . 1 <--VariableArray
5 --->> Waking KellerAcculevel at modbus_0x01 ... <--VariableArray
Waking KellerAcculevel at modbus_0x01 <--SensorBase
   ... wake up uccess. <<--- 5 <--VariableArray
3 . 1 --->> Collected result of reading 1 from InsituTrollSdi12 at SDI12-1_Pin7 ... <--VariableArray
Putting 0.11 in result array for variable 0 from InsituTrollSdi12 at SDI12-1_Pin7 <--SensorBase
+Putting 16.62 in result array for variable 1 from InsituTrollSdi12 at SDI12-1_Pin7 <--SensorBase
+Putting 0.24 in result array for variable 2 from InsituTrollSdi12 at SDI12-1_Pin7 <--SensorBase
+
   ... got measurement result. <<--- 3 . 1 <--VariableArray
3 . 2 --->> Starting reading 2 on InsituTrollSdi12 at SDI12-1_Pin7 ... <--VariableArray
Starting measurement on InsituTrollSdi12 at SDI12-1_Pin7 <--SensorBase
   ... set up succeeded. <<--- 3 . 2 <--VariableArray
3 . 2 --->> Collected result of reading 2 from InsituTrollSdi12 at SDI12-1_Pin7 ... <--VariableArray
Adding 0.11 to result array for variable 0 from InsituTrollSdi12 at SDI12-1_Pin7 <--SensorBase
+Adding 16.62 to result array for variable 1 from InsituTrollSdi12 at SDI12-1_Pin7 <--SensorBase
+Adding 0.24 to result array for variable 2 from InsituTrollSdi12 at SDI12-1_Pin7 <--SensorBase
+
   ... got measurement result. <<--- 3 . 2 <--VariableArray

SRGDamia1, 2021-02-11:
I think I found it!!

Here:  https://github.com/EnviroDIY/ModularSensors/commit/28b58ca02ae24c9a91516faff7f004d46ee60234

aufdenkampe, 2021-02-11:
@SRGDamia1 and @neilh10, thanks for working on this!

neilh10, 2021-02-11:
@SRGDamia1 thanks - just saw comment above- for some reason missed it. Will try testing.

SRGDamia1, 2021-02-11:
@neilh10 There were two places in the newer SDI-12 code where I was reading the value when debugging but leaving it in the buffer if debugging was off.  So, if you turned on debugging, it always worked fine.  But if you didn't turn on the debugging, there were characters being left in the buffer so one of the `while` loops was hanging until the watch-dog bit.   As you suggested, I went through the SDI-12 code writing comments for almost every line, which showed the problems.

SRGDamia1, 2021-02-11:
I didn't find it any of the first dozen or so tests because in order to hang you had to both have a sensor attached and have the debugging off.

neilh10, 2021-02-15:
@SRGDamia1 thanks for the insights.  It works for me with the LT500.  I only have one SDI-12 sensor the LT500, and only one specific unit of the LT500 to test against (its expensive but been reliable, and has its own built in logger).

From the new code, thanks for the extra inline descriptions, still working it through on paper, as the MS_DEBUG seems to upset the LT500.

Protocol work is challenging, as much because its often impossible(!!!)  to cover all the cases with varieties of equipment.  
The new code makes me  nervous as it has built in timeouts - with what I call banana skins - waiting to be slipped on.  
1) they can't be tested that I cans see off under standard test conditions
2) if they appear in "the field", with willing eyes watching or monitoring, they go unseen.

I've introduced the following on my fork 
```
        #define TIMEOUT_SDI12PRTCL_NOHEADER_MS 1500
        #define TIMEOUT_SDI12PRTCL_NOBODY_MS 3000
        while (_SDI12Internal.available() < 3 ) {
            if((millis() - start) > TIMEOUT_SDI12PRTCL_NOHEADER_MS) {
                PRINTOUT(F("SDI12Sensors::getResults1 ERROR waiting for response"));
                //contrinue on - likely to be other cascading failure msgs
                break;
            }
        }
```
and similarly 
```
        while (_SDI12Internal.available()) {
             if( (millis() - start) > TIMEOUT_SDI12PRTCL_NOBODY_MS) {
                PRINTOUT(F("SDI12Sensors::getResults2 ERROR sensor took too long to send data"));
                break;
             }
```

closed 2021-02-15 by neilh10