EnviroDIY/ModularSensors #350: platformio.ini file on Developer Setup contains a typo

mbarneytu, 2021-01-15:
On this page: https://envirodiy.github.io/ModularSensors/page_for_developers.html

The example platformio.ini, when pasted into a new PIO project, produces the following error in VSCode when doing a Clean:

`Error: Invalid 'C:\Users\Matt.Barney\Documents\Projects\PlatformIO\test0807\platformio.ini' (project configuration file): 'Source contains parsing errors: 'C:\\Users\\Matt.Barney\\Documents\\Projects\\PlatformIO\\test0807\\platformio.ini'
        [line 16]: 'EnviroDIY_DS3231\n''`

Apparently PIO wants whitespace in front of each line of a multi-line lib_deps list, and the first line of the example doesn't have that. If I insert a space or a tab in front of that first line, it fixes the error.

mbarneytu, 2021-01-15:
Another typo here:
`lib_ldf_mode = deep++`
should be deep+

mbarneytu, 2021-01-15:
The ini file specifies this in lib_deps: `Arduino-SDI-12`, which doesn't exist. I'm thinking it should just be `SDI-12`.

@SRGDamia1 I'm sorry for all these add-on comments to this issue; I swear I'm not trying to pile on. :slightly_smiling_face:  I'm just working through this as a first-time (potential) contributor to ModularSensors and reporting issues as I encounter them.

SRGDamia1, 2021-01-18:
I really appreciate the issue finding!

I think all of these things are fixed on the update_docs branch, but it hasn't made it to master yet.

closed 2021-01-18 by SRGDamia1

aufdenkampe, 2021-01-18:
Yes, thanks to both of you for finding and fixing issues!

mbarneytu, 2021-01-19:
I confirmed that the revised ini file in the [update_docs branch](https://github.com/EnviroDIY/ModularSensors/blob/update_docs/docs/FAQ/Developer-Setup.md)  fixes the issues noted above. However, when I use it to build for mayfly, I get compile errors about RTCZero, beginning with: 
```
In file included from .pio\libdeps\mayfly\RTCZero\src\RTCZero.cpp:22:0:
.pio\libdeps\mayfly\RTCZero\src\RTCZero.h:32:26: error: 'RTC_MODE2_MASK_SEL_OFF_Val' was not declared in this scope
     MATCH_OFF          = RTC_MODE2_MASK_SEL_OFF_Val,          // Never
```

even though RTCZero is listed in lib_ignore in the [env:mayfly] section.

If I remove arduino-libraries/RTCZero from the [env] section's lib_deps, it fixes these errors, and I can build successfully for both mayfly and adafruit_feather_m0 environments.

SRGDamia1, 2021-01-19:
Huh.  That's weird.  I don't get that error.  I'm using pio 5.0.4.

mbarneytu, 2021-01-19:
Weird. I'm also on 5.0.4. I had PIO create a new project, pasted in the example .ini, then built it for Mayfly, and it failed. It will build successfully for the adafruit_feather_m0. If I comment-out or delete RTCZero from the common lib_deps, it builds successfully for both environments. Mayfly build log is attached.
[mayfly build log.txt](https://github.com/EnviroDIY/ModularSensors/files/5838600/mayfly.build.log.txt)