EnviroDIY/ModularSensors #353: LTE-M1: Bluetooth, Cellular 4G LTE CAT-M1 (AT&T/Verizon) Availability

everest923, 2021-01-15:
Is the LTE-M1: Bluetooth, Cellular 4G LTE CAT-M1 (AT&T/Verizon) available anymore? If not is there an acceptable replacement? Is the Digi XBee Cellular LTE Cat 1, Verizon LTE Cat 1, Dual U.FL, TH, USA a suitable replacement?

Thanks!

mbarneytu, 2021-01-15:
To my knowledge, it is still available, and is listed at DigiKey.com. I recommend asking this type of question on [EnviroDIY's Mayfly forum](https://www.envirodiy.org/forum/mayfly-data-logger/); GitHub issues here are generally intended for feedback and bug reporting for the ModularSensors library code itself.
Best, Matt

neilh10, 2021-01-15:
I've bought these from Digikey.com 

602-2169-ND  XB3-C-A2-UT-001
1998-251-ND    VERIZON SIM TRIPLE FF CAT-M
https://dataplans.digikey.com/   there are now 50Mbyte/month plans
 requires the LTE Adapter

I wonder if the "discussions" could be enabled  ~  https://docs.github.com/en/discussions/quickstart

SRGDamia1, 2021-01-18:
The discussion feature looks interesting, but for now, I think I'd prefer to stick to the forums.

closed 2021-01-18 by SRGDamia1