EnviroDIY/ModularSensors #358: SDI-12 lib source differences

neilh10, 2021-02-16:
The git managed source control is core to being able to build reliable loads.  ModularSensors is absolutely amazing managing how they are pulled together.

I still figuring out the different mechanisms, and I might be missing something, but 
I'm getting a difference between what I pull directly and what comes from using the built in platformio build mechanism. 

Direct pull from 
https://github.com/EnviroDIY/Arduino-SDI-12.git  
which seems to be tagged v2.1.1
but has its library.json edited to 
"version": "2.1.2",

When it is pulled as part of a release in building from examples\menu_a_la_carte\.pio\libdeps\mayfly\SDI-12

the
```
.pio/libdeps/mayfly/SDI-12/library.json 
{
"name": "SDI-12",
"version": "2.1.2",
..
"repository": {
"type": "git",
"url": "https://github.com/EnviroDIY/Arduino-SDI-12.git"
},
```
seems to be saying it should come from the same place as direct pull, but they are different. 
The differences are not major, but the version by the release mechanism v2.1.2 has some relevant fixes. 

I was expecting they would be the same, but they are different.

SRGDamia1, 2021-02-16:
I will look into it.  If I remember correctly there was something strange going on with the PlatformIO registry and it wouldn't pick up the update.

neilh10, 2021-04-06:
This has been fixed for me. It now pulls v2.1.3 consistently.
However v2.1.3 has timing changes that have stopped recognizing the LT500/SDI-12 response which I'm investigating - https://github.com/neilh10/ModularSensors/issues/32

closed 2021-04-06 by neilh10