EnviroDIY/ModularSensors #381: STM32 Support

TzionCastillo1, 2021-07-23:
I'm looking to port my Modular Sensors based code over to an Arduino Portenta H7, which has a STM32H7 cpu. Is there a starting point you'd recommend for adding to the library to make it work for this processor as well?
Thanks

neilh10, 2021-07-26:
Wow – fantastic,  Portentia says it can reduce to 3uA with only RTC on,  4UARTS, and LiPo charge to 1.5A
https://search.arduino.cc/search?tab=reference&q=Portenta+
though looking at the data sheet not clear to me exactly what the “Default Configuration is”. Seems like it does  include a local data flash but only the options are described not the specific flash size.  The uSD is off board, possibly on BreakOut board for prototyping.
https://content.arduino.cc/assets/Datasheet-Portenta-H7.pdf
I’ve been working on a Wio Terminal SAMD51 port  M4+Display+WiFi/BLE
What I’ve done is first verified that the Wio Terminal WiFi was stable ~ the initial release of Seeed software wasn’t stable.  I’m still in process of testing the WiFi – though do have rudimentary NTP/UDP working. 
So partly depends on your target, assuming you are thinking M4,  I would suggest
1)	A basic logger recording to (local flash or uSD). The mayfly 0.5 supports ext uSD, but this might be being extended.
2)	Derive wall time – possibly in setup() Wifi NTP/UDP, but then look to extend LoggerModem Class to support the specific WiFi
3)	I assume you want to push to a cloud – but which one. monitormywatershed.org  / WiFi is a good test reference. You would just need to get the underlying WiFi connectivity.
I did pull up a the PlatformIo Portenta H7 M4 processor on Arduino and am empty framework compiled OK. But the basic framework is  a lot of ram 
Checking size .pio\build\portenta_h7_m4\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.7% (used 55016 bytes from 294248 bytes) 
Flash: [=         ]   8.6% (used 90368 bytes from 1048576 bytes)

I wonder if you could share a little on the app – what sensors interfacing to?, and remote cloud? And power source?.

TzionCastillo1, 2021-07-27:
Yeah no problem. I am using the Atlas Scientific suite of sensors as well as an Adafruit GPS sensor(currently just as a calculated variable since it is unsupported), am publishing to Thingspeak, and have it powered through a Li-Po battery. I currently have this configuration working on the Mayfly board, but I need to implement an ML model into my project, and figured the dual-core Portenta would prove to be a more elegant solution rather than using a Raspberry  in parallel with the Mayfly.

TzionCastillo1, 2021-07-27:
The glaring issues I've ran into so far have to do with he processor itself. I am targeting the M4 core, so I am running into problems with the EnableInterrupts library. Are you aware of a work around for this?

neilh10, 2021-07-27:
Nice spec H7-15EUCWAD. I've been chewing on some possible ML, but I haven't graduated from playing with rPi and was thinking of the STM32MP157C.
the Mayfly uses the external RTC to generate its interrupt. 
Its got to be switched to the internal clock ~ for SAMDxx internal clock its a total different interface.
If you do a pull  against my https://github.com/neilh10/ModularSensors (default is branch/stream release1) 
and compare against the base, I have slightly expanded it.

However it did give me a whole bunch of trouble with the M4 see
LoggerBase.cpp - 1043 #elif defined ARDUINO_ARCH_SAMD

The time aspects are pretty confusing, and its a question how to present them??,
https://github.com/EnviroDIY/ModularSensors/issues/310
My take is anything that is user is local time - uSD, and debug. (summer time adjust is toooooo changeable).
Interfacing to the world is epoch UTC ~ 1970
SAMD RTC is yr 2k  based
DS3231 is effectively 2k  ( I changed the ref to rtcExtphy)

default time is based on the compile time-aday I think, since I'm in Ca UTC-8 - though  generaly https://github.com/EnviroDIY/ModularSensors/issues/304
A lot of time init is in magic numbers, and I've used a #define to be clear about.
other utilities need to be characterized, some are based on 1900 
The Mayfly DS3231 also incorporates a DateTime  class accepts initialization in yr 2k.  
For SAMD I switched to a different standard Adafruit https://github.com/neilh10/RTCZero with modifications to keep it compatible with some of the DS3231 DataTime.

My protoyping  port was using an Adafruit SAMD51, which doesn't have battery backup. I prototyped a separate low power rtcExtPhy to always be able to boot with a time.
My target alpha development build is under ModularSensors\a\kn_depth  and is probably broken right now as haven't touched for some time.
So got complex, hope that helps.

SRGDamia1, 2021-11-24:
All of the processor-dependent stuff should be in LoggerBase and WatchDogs, with some sneaky kickers probably hiding in the SDI-12's interrupt dependency.  

Like Neil's hinting at, the big things would be to get the sleep/wake and working with the interrupts.  Figuring out the watchdog alarm can also be really hard, but if you're willing to just spoof out that feature, it's a lot easier because it's broken into a separate module.  It's on my wish list to find time to split the clock and sleep functionality out similarly to the watchdogs, but there's very, very little chance that will actually happen in the very near future.