EnviroDIY/ModularSensors #382: Dependency Issues

TzionCastillo1, 2021-08-17:
My Laptop was stolen a few weeks ago, so I had to move to a different machine for development, but same OS, and still using Platform.io on VSCode. However, trying to build my previously functioning code on my new computer, I run into dependency issues such as : 

[image]

But I get this error with several of the libraries that are required for Modular Sensors to compile ie. DHT, AM2315, ADS1015, etc.
I get these errors both with my previously working code, as well as example code from Github, even when in a new project.

Any help would be much appreciated

fisherba, 2021-08-17:
This may not be the best answer, but I have resolved this by deleting the .pio directory and letting PlatformIO completely rebuild the libraries.

TzionCastillo1, 2021-08-17:
I attempted this but I am getting a similar error:
[image]

neilh10, 2021-08-18:
Ouch your last screen looks like a Python error. Seems to be working, and then suddenly can't find the python file.

Mostly PlatformIO has been fantastic, but because of changes in Visual Studio Code and Windows 10 protection I've occasionally run into problems,
You could try looking for the problem on https://community.platformio.org/

You could try using the CLI tools to ensure its installed right. That worked for me when I had a problem recently, got it fixed, but was torturous and not sure what went wrong that I then had to fix.

See scripts in <user>\.platformio\penv\Scripts\pio

https://docs.platformio.org/en/latest/core/installation.html#piocore-install-shell-commands

From outside VSC I ended up doing something like this 

$ pip uninstall platformio   
$ pip install -U platformio
It then rebuilt all the tool

GArrigotti-cws, 2021-08-18:
I'd be hesitant to uninstall PlatformIO directly through your Python command line, to ensure any virtual environments are not hindered I would simply uninstall PlatformIO in Visual Studio Code and reinstall the software.  That should configure the virtual environment to Visual Studio Code.  Then if the error occurs with Modular Sensors I would run the "pio run -t clean" command or hit the button, it should rebuild the indexes correctly.

TzionCastillo1, 2021-08-20:
After some searching, I found out that this error is cause by Git not being installed globally. Installed Git on my new machine and everything is working now.

Thanks for all your help!

closed 2021-08-20 by TzionCastillo1