EnviroDIY/ModularSensors #412: Case collision in git Windows

neilh10, 2022-05-05:
When changing a case on a file, it has problems with the Windows  environment.
The case was changed from InsituTrollSdi12a.h  to InSituTrollSdi12a.h

@SRGDamia1  Just wondering how you solved it across the branches when you changed the case. Many thanks

Cloning into '<>\.platformio\.cache\tmp\pkg-installing-wsc30bhb'...
warning: the following paths have collided (e.g. case-sensitive paths
on a case-insensitive filesystem) and only one from the same
colliding group is in the working tree:

  'src/sensors/InSituTrollSdi12a.h'
  'src/sensors/InsituTrollSdi12a.h'

neilh10, 2022-05-18:
well just wondering if anybody else is seeing this.
I'm still on 0.33.1, and possibly this will go away when I merge to the next, but I think its in the windows directory.
If it doesn't, it seems to me the only solution is to start a new repository in my windows directory so there aren't turds left behind. But maybe somebody has a simpler solution.

aufdenkampe, 2022-05-19:
@neilh10, I've been using v0.33.x without any problems.

The one exception are the issues with PlatormIO Core 6.0.0 that happened to everyone yesterday and over the weekend. See https://github.com/platformio/platformio-core/issues/4270. If didn't use it since Friday, you won't have these problems, as the 6.0.1 fix was released late on Monday.

@neilh10, I'm actually quite perplexed at how it is possible that you are getting the error the you describe if you use PlatformIO in the recommended way. You've made comments in the past that have also led me to wonder if you are setting up PlatformIO properly. Maybe we should do a screen share and you could walk me through your issue. I don't think it is an issue with ModularSensors.

neilh10, 2023-06-10:
Well thanks for the comments - its actually a pure git issue  - and seems like it is only my fork of git. Somehow I've got on the bleeding edge

This just a heads up that Windows treats case different than linux (which git was built in) and that when changing a file name - please don't just change the case in a file and then push it as in 
 'src/sensors/InSituTrollSdi12a.h'
 'src/sensors/InsituTrollSdi12a.h'
This is described elsewhere 
https://github.com/clearlydefined/curated-data/issues/7837
And a solution described here is hard work
https://stackoverflow.com/questions/59643532/git-clone-error-under-windows-with-symlinks

When I clone a new git in my fork - it hasn't resolved the case change introduced into the (main)
Probably because I have the other name still on a branch.

$ git clone git@github.com:neilh10/ModularSensors.git
Cloning into 'ModularSensors'...
remote: Enumerating objects: 35825, done.
remote: Counting objects: 100% (7411/7411), done.
remote: Compressing objects: 100% (1738/1738), done.
remote: Total 35825 (delta 5904), reused 7059 (delta 5620), pack-reused 28414
Receiving objects: 100% (35825/35825), 159.62 MiB | 2.98 MiB/s, done.
Resolving deltas: 100% (27478/27478), done.
warning: the following paths have collided (e.g. case-sensitive paths
on a case-insensitive filesystem) and only one from the same
colliding group is in the working tree:

  'src/sensors/InSituTrollSdi12a.h'
  'src/sensors/InsituTrollSdi12a.h'

However when it I clone directly there isn't the problem 
$ git clone git@github.com:EnviroDIY/ModularSensors.git
Cloning into 'ModularSensors'...
remote: Enumerating objects: 30139, done.
remote: Counting objects: 100% (6029/6029), done.
remote: Compressing objects: 100% (1254/1254), done.
remote: Total 30139 (delta 5016), reused 5615 (delta 4731), pack-reused 24110
Receiving objects: 100% (30139/30139), 164.62 MiB | 3.11 MiB/s, done.
Resolving deltas: 100% (23506/23506), done.

$ git clone git@github.com:mbarneytu/ModularSensors.git
Cloning into 'ModularSensors'...
remote: Enumerating objects: 24670, done.
remote: Counting objects: 100% (561/561), done.
remote: Compressing objects: 100% (66/66), done.
remote: Total 24670 (delta 529), reused 509 (delta 493), pack-reused 24109
Receiving objects: 100% (24670/24670), 158.15 MiB | 2.84 MiB/s, done.
Resolving deltas: 100% (19019/19019), done.

I'm leaving this open for info