EnviroDIY/ModularSensors #415: RTC options - discussion

neilh10, 2022-06-15:
This is a discussion for working with the myrad RTC hardware and interface library's and see if anybody has some ideas or solutions.

“current time” is an essential part of ModularSensors, and managed by a machine/platform specific source called a RealTimeClock (RTC). The Mayfly is tightly tied to the external RTC DS3231 to manage accurate time, and this device is accessed through a lib Sodaq_DS3231

When adapting to different systems/platforms (Adafruit Feather samd21/M0 and samd51/M4, Seeed Wio Terminal with WiFi/samd51/M4 and Espriff) - the RTC is usually different from that of the Mayfly and has different helper functions to derive “current time” 
ModularSensors/Mayfly is accessed through the lib Sodaq_DS3231 . 
The Sodaq_DS3231.h includes helper class DateTime – and DateTime is implemented across many environments with incompatible implementations.

A lot of newer processors support an internal hardware clock or hardware timer, however it needs to survive a power down, and this is usually supplied with an external RTC.
With standard running conditions, the internal hardware clock is used, but on reset and audit conditions the RTC is used as higher accuracy time source.

Different platform environments support “current time” in different ways. 
An objective for ModularSensors would be to abstract these hardware layers, however this is likely to have a large impact on the current code.

The standard solution implemented elsewhere, is to have a Hardware Abstraction Layer (HAL)  but this unlikely to fit with the current Arduinio framework very easly. It may be there with the next generation Arm compatible Arduino but I haven’t checked.

A future objective may be to support an extensible external RTC environment such as 
https://github.com/adafruit/RTClib/tree/master/src

For specific processors (SAMD21 SAMD51) the DateTime collision also happens between implementations of accessing the RTC timers  adafruit and seeed. 

Eg  for SAMD51 on Wio Terminal,  with WIFI  
    https://github.com/Seeed-Studio/Seeed_Arduino_RTC 
seeed_wio_terminal\Seeed Arduino RTC\src\DateTime.h has DateTime.cpp

Using namespace its possible to isolate DateTime in Sodaq_DS3231.h 
3 lines in Sodaq_DS3231.h and 1 line in Sodaq_DS3231
https://github.com/neilh10/Sodaq_DS3231/tree/rel1_dt_namespace

and this maybe possible to extend it to other processor modules with minor adaptions

Apologies if this isn't very clear,  but it takes working with specific environments to find out how its solved in those environments and then the small incompatible differences to see how much of a problem this is . So wondering if anybody has found some other working solutions.

SRGDamia1, 2022-06-16:
I'll look into the RTC library.

ModularSensors does support using the built-in RTC for the SAMD21.

SRGDamia1, 2024-05-16:
I also want to add support for the RV-8803: https://www.microcrystal.com/en/products/real-time-clock-rtc-modules/rv-8803-c7/

I suspect the Adafruit library may be the best bet, after adding the 8803 to it.