EnviroDIY/ModularSensors #416: Implement Campbell RainVUE10 SDI-12 Precipitation Sensor

aufdenkampe, 2022-06-20:
Implement ModularSensors wrapper for [Campbell RainVUE10 SDI-12 Precipitation Sensor with Plastic Funnel](https://www.campbellsci.com/rainvue10).

Using the following Campbell documentation
- Brochure: [RainVUE10 SDI-12 Precipitation Sensor](https://s.campbellsci.com/documents/us/product-brochures/b_rainvue10.pdf)
- Manual: [RainVUE-Series SDI-12 Precipitation Sensors](https://s.campbellsci.com/documents/us/manuals/rainvue-series.pdf)

Implementing within the [`RainVUE`](https://github.com/EnviroDIY/ModularSensors/tree/RainVUE) branch.

aufdenkampe, 2022-06-27:
I have this mostly working, but for some reason, ModularSensors isn't properly parsing the Get Measurement response (or something like that):

Here's what I get from Serial Monitor when running the [simple_logging_rainvue.ino](https://github.com/EnviroDIY/ModularSensors/blob/RainVUE/sensor_tests/simple_logging_rainvue/simple_logging_rainvue.ino) test sketch with `MS_SDI12SENSORS_DEBUG` and `MS_SDI12SENSORS_DEBUG_DEEP` defined in the .ini:

```
Setting up sensors...
Enabling interrupts for SDI12 on pin 5 <--SDI12Sensors
  Asking for sensor acknowlegement <--SDI12Sensors
    >>> 0! <--SDI12Sensors
    <<< ␀n <--SDI12Sensors
    Campbell RainVUE10 at SDI12-0_Pin5 did not reply! <--SDI12Sensors
    >>> 0! <--SDI12Sensors
    <<< 0 <--SDI12Sensors
    Campbell RainVUE10 at SDI12-0_Pin5 replied as expected. <--SDI12Sensors
  Getting sensor info <--SDI12Sensors
    >>> 0I! <--SDI12Sensors
    <<< 014CampbellRV10IN200SN=210908 <--SDI12Sensors
   SDI12 Address: 0 <--SDI12Sensors
   SDI12 Version: 1.40 <--SDI12Sensors
   Sensor Vendor: Campbell <--SDI12Sensors
   Sensor Model: RV10IN <--SDI12Sensors
    Sensor Version: 200 <--SDI12Sensors
   Sensor Serial Number: SN=210908 <--SDI12Sensors
Data will be saved as XXXXX_2022-06-27.csv
------------------------------------------
  Asking for sensor acknowlegement <--SDI12Sensors
    >>> 0! <--SDI12Sensors
    <<< 0 <--SDI12Sensors
    Campbell RainVUE10 at SDI12-0_Pin5 replied as expected. <--SDI12Sensors
  Beginning concurrent measurement on Campbell RainVUE10 at SDI12-0_Pin5 <--SDI12Sensors
    >>> 0C! <--SDI12Sensors
    <<< 000005 <--SDI12Sensors
Campbell RainVUE10 at SDI12-0_Pin5 did not respond to measurement request! <--SDI12Sensors
Campbell RainVUE10 at SDI12-0_Pin5 is not currently measuring! <--SDI12Sensors

 \/---- Line Saved to SD Card ----\/
2022-06-27 16:18:00,1,13272,4.806,30.50,-9999.00,-9999,-9999.00,-9999.00
```

@SRGDamia1, any suggestions? I have much less experience with SDI-12 than I do with Modbus.

aufdenkampe, 2022-06-27:
I just tried also defining `MS_SDI12_NON_CONCURRENT` in the `.ini`, even though SDI12 Version: 1.40, just to see if there is a difference, but the response is nearly identical:

```
------------------------------------------
  Asking for sensor acknowlegement <--SDI12Sensors
    >>> 0! <--SDI12Sensors
    <<< 0 <--SDI12Sensors
    Campbell RainVUE10 at SDI12-0_Pin5 replied as expected. <--SDI12Sensors
  Beginning NON-concurrent (standard) measurement on Campbell RainVUE10 at SDI12-0_Pin5 <--SDI12Sensors
    >>> 0M! <--SDI12Sensors
    <<< 00005 <--SDI12Sensors
Campbell RainVUE10 at SDI12-0_Pin5 is not currently measuring! <--SDI12Sensors

 \/---- Line Saved to SD Card ----\/
2022-06-27 16:29:00,3,13272,4.821,31.75,-9999.00,-9999,-9999.00,-9999.00
```

aufdenkampe, 2022-06-27:
Here's my output from [/EnviroDIY/Arduino-SDI-12/examples/d_simple_logger](https://github.com/EnviroDIY/Arduino-SDI-12/tree/master/examples/d_simple_logger)"

```
Opening SDI-12 bus...
Timeout value: 
-9999
Powering up sensors...
Scanning all addresses, please wait...
Sensor Address, Protocol Version, Sensor Vendor, Sensor Model, Sensor Version, Sensor ID
, 0.00, , , , , 
Total number of sensors found:  1

Time Elapsed (s), Sensor Address, Est Measurement Time (s), Number Measurements, Real Measurement Time (ms), Measurement 1, Measurement 2, ... etc.
-------------------------------------------------------------------------------
24543, 0, 0, 5, 0.0000000000, 0.0000000000, 0.0000000000, 0.0000000000, 0.0000000000
25903, 0, 1, 5, 416, 3.0449224000, 28.2500020000, 1.7614121000, 0.0003010000, 1.7130642000
27013, 0, 0, 0, 
38178, 0, 0, 5, 0.0000000000, 0.0000000000, 0.0000000000, 0.0000000000, 0.0000000000
39532, 0, 1, 5, 416, 3.0449224000, 28.2500020000, 1.7468262000, 0.0004630001, 2.0374441000
40642, 0, 0, 0, 
51806, 0, 0, 5, 0.0352540050, 3.0000000000, 0.0352540050, 25.3013920000, 39.5648230000
53463, 0, 1, 5, 413, 3.0439463000, 28.2500020000, 1.7453752000, 0.0000460000, 2.2037032000
54573, 0, 0, 0, 
65736, 0, 0, 5, 0.0000000000, 0.0000000000, 0.0352540050, 0.0000000000, 0.0000000000
67141, 0, 1, 5, 416, 3.0371094000, 28.2500020000, 1.7468262000, 0.0001970000, 2.2645502000
68237, 0, 0, 0, 
79405, 0, 0, 5, 0.0000000000, 0.0000000000, 0.0352540050, 0.0000000000, 0.0000000000
80812, 0, 1, 5, 415, 3.0351563000, 28.2500020000, 1.7595361000, 0.0003590000, 2.2659752000
81915, 0, 0, 0, 
93085, 0, 0, 5, 0.0000000000, 0.0000000000, 0.0352540050, 0.0000000000, 0.0000000000
94490, 0, 1, 5, 416, 3.0400393000, 28.1875020000, 1.7732582000, 0.0005320000, 2.2645502000
95598, 0, 0, 0, 
106764, 0, 0, 5, 0.0441260040, 4.0000000000, 0.0793800060, 10.8442310000, 14.3359210000
108414, 0, 1, 5, 413, 3.0078132000, 28.1875020000, 1.7326772000, 0.0000580000, 2.2364674000
```

aufdenkampe, 2022-06-27:
Similar to above ([/EnviroDIY/Arduino-SDI-12/examples/d_simple_logger](https://github.com/EnviroDIY/Arduino-SDI-12/tree/master/examples/d_simple_logger)), but set to run SDI-12 from Mayfly pin 5 (constant 3.3V).  Note that this immediately finds the sensor info, whereas it did not find when on switched power.

```
Opening SDI-12 bus...
Timeout value: 
-9999
Powering up sensors...
Scanning all addresses, please wait...
Sensor Address, Protocol Version, Sensor Vendor, Sensor Model, Sensor Version, Sensor ID
0, 1.40, Campbell, RV10IN, 200, SN=210908, 
Total number of sensors found:  1

Time Elapsed (s), Sensor Address, Est Measurement Time (s), Number Measurements, Real Measurement Time (ms), Measurement 1, Measurement 2, ... etc.
-------------------------------------------------------------------------------
24559, 0, 0, 5, 0.0000000000, 0.0000000000, 0.0000000000, 0.0000000000, 0.0000000000
25913, 0, 1, 5, 416, 2.9863284000, 26.6250020000, 1.6428492000, 0.0009610001, 2.2307692000
27029, 0, 0, 0, 
38193, 0, 0, 5, 0.0226420030, 2.0000000000, 0.0226420030, 19.1626210000, 37.9431110000
39849, 0, 1, 5, 413, 3.0214844000, 26.6875020000, 1.7017082000, 0.0000350000, 2.1739922000
40974, 0, 0, 0, 
52142, 0, 0, 5, 0.0102120010, 1.0000000000, 0.0328540020, 2.7002072000, 2.7002072000
53800, 0, 1, 5, 413, 3.0166023000, 26.6250020000, 1.6569031000, 0.0000460000, 2.2234433000
54923, 0, 0, 0, 
66088, 0, 0, 5, 0.0000000000, 0.0000000000, 0.0328540020, 0.0000000000, 0.0000000000
67495, 0, 1, 5, 415, 3.0146494000, 26.5625020000, 1.6428492000, 0.0002080000, 2.2419612000

```

aufdenkampe, 2022-06-28:
I've played around with Warm Up and Measurement times, from 0 to 10,000 ms, and tried a 30 ms Extra Wake time.

I've tried concurrent and non-concurrent, and right now I'm playing around with modifying the  `C!` and  `M!` commands in the `SDI12Sensors.cpp` to alternatives provided in the sensor manual.

None work.

aufdenkampe, 2022-06-28:
I now have some results, but not yet the ones I need.

I changed the `C!` command to `C1!` in  `SDI12Sensors.cpp`, which giving me the following instrument diagnostic info:
1. Supplyvoltage
2. Temperature
3. Sensorlevel
4. Timesincelastmeasurable precipitation event
5. Lithiumbatteryvoltage

Here's the serial monitor response.

```
------------------------------------------
  Asking for sensor acknowlegement <--SDI12Sensors
    >>> 0! <--SDI12Sensors
    <<< 0 <--SDI12Sensors
    Campbell RainVUE10 at SDI12-0_Pin5 replied as expected. <--SDI12Sensors
  Beginning concurrent measurement on Campbell RainVUE10 at SDI12-0_Pin5 <--SDI12Sensors
    >>> 0C1! <--SDI12Sensors
    <<< 000105 <--SDI12Sensors
    Concurrent measurement started. <--SDI12Sensors
Campbell RainVUE10 at SDI12-0_Pin5 is reporting: <--SDI12Sensors
    >>> 0D0! <--SDI12Sensors
    <<< 0 <--SDI12Sensors
    <<< + <--SDI12Sensors
    <<< 3.0136724000 <--SDI12Sensors
    <<< + <--SDI12Sensors
    <<< 23.5000000000 <--SDI12Sensors
    <<< + <--SDI12Sensors
    <<< 1.7890102000 <--SDI12Sensors
    <<< + <--SDI12Sensors
    <<< 0.0290970040 <--SDI12Sensors
    <<< + <--SDI12Sensors
    <<< 2.2623112000 <--SDI12Sensors
 <--SDI12Sensors
    <<< 
 <--SDI12Sensors
  Total Results Received:  5 , Remaining:  0 <--SDI12Sensors

 \/---- Line Saved to SD Card ----\/
2022-06-28 07:00:00,5,13293,4.791,25.25,3.01,23,0.03,2.26
```

aufdenkampe, 2022-06-28:
@gcutrell, do you have any suggestions on what I might try?

SRGDamia1, 2022-06-28:
Does example k in the SDI-12 library work?

aufdenkampe, 2022-06-28:
@SRGDamia1, thanks for the tip!

Example K does not work. Here is the serial monitor response:

```
pening SDI-12 bus...
Timeout value: 
-9999
Powering up sensors...
Scanning all addresses, please wait...
Protocol Version, Sensor Address, Sensor Vendor, Sensor Model, Sensor Version, Sensor ID
Total number of sensors found:  0
No sensors found, please check connections and restart the Arduino.
```

gcutrell, 2022-06-28:
@aufdenkampe @SRGDamia1 I think the issue is in SDI12Sensors.cpp, the program doesn't allow for a zero wait time. I recently discovered this and just added the code below before "if (wait > 0) {" in the concurrent and non-concurrent sections. 
    //It is possible for wait time to equal 0. To allow program to continue set wait time to 1
    if (wait ==0) wait = 1;

aufdenkampe, 2022-06-28:
@gcutrell, that did it! Thanks!!!

```
------------------------------------------
  Asking for sensor acknowlegement <--SDI12Sensors
    >>> 0! <--SDI12Sensors
    <<< 0 <--SDI12Sensors
    Campbell RainVUE10 at SDI12-0_Pin5 replied as expected. <--SDI12Sensors
  Beginning concurrent measurement on Campbell RainVUE10 at SDI12-0_Pin5 <--SDI12Sensors
    >>> 0C! <--SDI12Sensors
    <<< 000005 <--SDI12Sensors
    Concurrent measurement started. <--SDI12Sensors
Campbell RainVUE10 at SDI12-0_Pin5 is reporting: <--SDI12Sensors
    >>> 0D0! <--SDI12Sensors
    <<< 0 <--SDI12Sensors
    <<< + <--SDI12Sensors
    <<< 0.0212590020 <--SDI12Sensors
    <<< + <--SDI12Sensors
    <<< 2.0000000000 <--SDI12Sensors
    <<< + <--SDI12Sensors
    <<< 0.0212590020 <--SDI12Sensors
    <<< + <--SDI12Sensors
    <<< 6.3817167000 <--SDI12Sensors
    <<< + <--SDI12Sensors
    <<< 12.7521310000 <--SDI12Sensors
 <--SDI12Sensors
    <<< 
 <--SDI12Sensors
  Total Results Received:  5 , Remaining:  0 <--SDI12Sensors

 \/---- Line Saved to SD Card ----\/
2022-06-28 09:27:00,2,13268,4.791,26.50,0.02,2,6.38,12.75
```

SRGDamia1, 2022-06-28:
Sorry, sloppy on my part.  In the `startSDI12Measurement` the wait is initialized as zero, with the assumption that if the sensor returns a value, it will be set as something other than 0.  It would be better to fix the problem in that function rather than in the `startSingleMeasurement` functions.

03e45afc70e1d3c53bc2486a9bd3fce1fa6abf77

aufdenkampe, 2022-06-28:
@SRGDamia1, thanks for a cleaner solution!

One last interesting thing: all this only works when I define `MS_SDI12SENSORS_DEBUG_DEEP`!

If I turn off debugging flags, or only define `MS_SDI12SENSORS_DEBUG`, it still doesn't work!

I'm taking off the rest of the week, so I won't be able to sleuth out the cause of this until mid-next week at the earliest.

SRGDamia1, 2022-06-29:
There was a bug in the non-concurrent measurement that was making it freeze without the debugging. I found it last week when Shannon needed to deploy some really old 5TM soil sensors. That should be fixed in one of the commits I pushed to your branch from master.

aufdenkampe, 2022-08-02:
@SRGDamia1, I forgot to follow up and let you know that it all worked, and the station was deployed a couple of days later. Data have been streaming nicely (with a few -9999 values here and there...).  See https://monitormywatershed.org/sites/WOODB-Well18/ 

I'll issue a PR to `develop` to track progress on cleanup and pulling this in.

closed 2022-08-02 by aufdenkampe