EnviroDIY/ModularSensors #417: [Newbie Question] Can I use this library to directly grab data from my TEROS-12 sensor?

lolcocks123, 2022-06-30:
Hello,

I am a complete Arduino newbie.

I was wondering if this library can be used with just an Arduino Uno and a TEROS-12  sensor to directly grab data from the sensor?

Regards,
Lolcocks.

SRGDamia1, 2022-07-01:
Are you trying to deploy a monitoring station or only running tests on the bench?  

If you want to quickly view data, use [example D on the SDI-12 library](https://github.com/EnviroDIY/Arduino-SDI-12/tree/master/examples/d_simple_logger).  This library is overkill.

If you're trying to create a long-term logging station, this library is the way to go, but I don't recommend an Uno for long deployments.

lolcocks123, 2022-07-01:
Thank you so much @SRGDamia1, I will start with the simple data view first and then come back to this library if required.

I am really grateful for your help.

closed 2022-07-01 by lolcocks123