EnviroDIY/ModularSensors #425: Visible (flashing light) warning when failed to write to SD card

SRGDamia1, 2022-11-14:
Add a flashing light on failure

neilh10, 2022-11-14:
Just a thought, if considering how a field technician (non programmer) without a computer, is using or visiting the site, it might be good to have a clear LED sequence 
Clearly somebody installing/maintaining the site needs an indication its  WORKING or NOT working, and there needs to be general power saving.

Other meaningful LEDs for MAYFLY 1.1 the Switched Power Out  applied RED led 
Network Bee Status - blue
Bee Power - white 

Some ideas here https://github.com/neilh10/ModularSensors/issues/3

GArrigotti-cws, 2022-11-14:
That’s a great idea, we have non programmers maintaining our sites.