EnviroDIY/ModularSensors #455: Disappeared github.com/MartinL

neilh10, 2023-09-28:
The following lib has disappeared... poof... gone as of this time.
 [github.com/MartinL1/BMP388_DEV.git](https://github.com/MartinL1/BMP388_DEV.git)
It is specified in library.json

It is available through 
https://registry.platformio.org/libraries/martinl1/BMP388_DEV

@SRGDamia1  you look like you have put a lot of thought into build management ( library.json ) and have some great build options for tracking build dependencies and are utilizing registry.platformio.org - just wondering what your thoughts are for managing dependencies
many thanks

mbarneytu, 2023-10-01:
I'm seeing the same issue. ModularSensors requires BMP388_DEV, but it is no longer available on github at the location specified by ModularSensors' library.json file.

Note that Arduino IDE's Library Manager is able to successfully install BMP388_DEV, but PlatformIO is not. 

As a temporary workaround, I downloaded the ModularSensors_Dependencies zip file from [the Releases folder](https://github.com/EnviroDIY/ModularSensors/releases) and copied the libraries it contains into my PlatformIO project folder:
...\.pio\libdeps\mayfly

SRGDamia1, 2023-10-02:
Those are the two best workarounds for now. I'll look into an alternate library for the BMP.

aufdenkampe, 2023-10-25:
BTW, I just created a clone here: https://github.com/EnviroDIY/BMP388_DEV

@SRGDamia1, thanks for your fix to `library.json` with commit db7bfb0d3af7bfbcc42c16a236dc53c440c7529b.

Can you issue a minor bug fix release so that people who build with PIO using 
`lib_deps = envirodiy/EnviroDIY_ModularSensors` can get the fix?

neilh10, 2023-10-25:
Hey thanks so much for the fixes. I see it when I pull in (master)

aufdenkampe, 2024-05-01:
@SRGDamia1, the PlatformIO Library Registry still has this bug, because we haven't issued an official release for a year.
https://registry.platformio.org/libraries/envirodiy/EnviroDIY_ModularSensors

We've had 23 commits to `master` since then.

Can you fix the failing tests on our `master` branch and issue an new release soon?

aufdenkampe, 2024-05-21:
Fixed with release v0.35.0!

closed 2024-05-21 by aufdenkampe