EnviroDIY/ModularSensors #456: Consider changing Adafruit_ADS1X15 dependency

aufdenkampe, 2023-11-22:
This is a key library for TI's popular Analog to Digital Converter (ADC) used in the Mayfly and many other boards.

Back in 2019 (or earlier), @SRGDamia1 decided to use this fork of Adafruit's library because it fixed some key bugs, and Adafruit wasn't accepting a pull requests to fix it (see https://github.com/EnviroDIY/ModularSensors/issues/235#issuecomment-470208394):
- https://github.com/soligen2010/Adafruit_ADS1X15

However, since then, Adafruit has renewed active development and issued a series of v2.x releases (which had breaking changes). 
- https://github.com/adafruit/Adafruit_ADS1X15

Adafruit's v2.4.2 is now [61 commits ahead of soligen2010](https://github.com/soligen2010/Adafruit_ADS1X15/compare/master...adafruit:Adafruit_ADS1X15:master), and is being actively pushed to PlatformIO's Library Registry:
- https://registry.platformio.org/libraries/adafruit/Adafruit%20ADS1X15

The major issue is that code developed for https://github.com/adafruit/Adafruit_ADS1X15 will often not work with ModularSensors because of the new and useful functions added Adafruit's v2 version.

@SRGDamia1, what would it take to switch back?

aufdenkampe, 2023-11-22:
@SRGDamia1, here are some of the new capabilities of [Adafruit's v2.4.2](https://github.com/adafruit/Adafruit_ADS1X15/releases) that would be handy (and are now being used by others):
- Add function to compute volts from ADC reading
- Add ability to set data rate
- Added a `bool` return type for `begin()` to allow checking initialization success.
- Adds ability to other differential channel configurations
- Support added for continuous reading. (see new continuous example)

Here are the breaking changes introduced to [v2.0.1](https://github.com/adafruit/Adafruit_ADS1X15/releases/tag/2.0.1):
- The library files name has been changed from `Adafruit_ADS1015` to `Adafruit_ADS1X15`. You will need to update `#include`s
- The I2C address is now passed in via `begin()` and not the c-tor. If you were using a non-default address, then you will need to make that change.
- The library now uses the `Adafruit_BusIO` library, which is a new dependency.
-

SRGDamia1, 2023-11-22:
It's been a while.  I know there was a reason I picked the fork, but I don't remember it now.

aufdenkampe, 2023-11-22:
With commit e2a9410 I just updated the entire repo to compile using Adafruit's version, so I think it should all work.

The comments that I could find (in https://github.com/EnviroDIY/ModularSensors/issues/235#issuecomment-470208394) was that there was a bug in Adafruit's version, which they didn't address for years. But that was 2016ish, and they really picked up development again starting in 2020. It's been quite active since, so I'm going to presume that bug was fixed.

Some of the features added by soligen2010, such as continuous mode (a0b81d5c4e0da0367d293e8610c825e5aee7f9c7) seem to have been added by Adafruit.

After a bit of research and work, I'm now strongly leaning toward making the switch, but it might make sense if you or someone at Stroud could test my branch a couple of your analog sensors.

aufdenkampe, 2023-11-27:
@SRGDamia1 & @s-hicks2, could you test that my [AlphasenseCO2](https://github.com/EnviroDIY/ModularSensors/tree/AlphasenseCO2) branch works with other sensors that use the ADS1X15 ADC?
- #457 shows my collective changes.
- Commit 87f4def lists the other sensors that need to be tested.

Thanks!

aufdenkampe, 2024-02-22:
@SRGDamia1 & @s-hicks2, can either of you test that my [AlphasenseCO2](https://github.com/EnviroDIY/ModularSensors/tree/AlphasenseCO2) branch works with other sensors that use the ADS1X15 ADC?

I really think this is an important and valuable update and that we should move forward with this Pull Request to the `develop` branch.
- #457

aufdenkampe, 2024-10-01:
@SRGDamia1, this issue can probably be closed given that you merged this PR:
- #457 

Have you tried the modified ADS1X15 dependency for any other sensors?

aufdenkampe, 2024-10-01:
Completed with Release [v0.36.0](https://github.com/EnviroDIY/ModularSensors/releases/tag/v0.36.0)!

closed 2024-10-01 by aufdenkampe